OrcaSlicer/OrcaSlicer #2540: Forced support - blocker and painter have no effect

tarsonis123, 2023-10-30:
### OrcaSlicer Version

1.8.0-beta1

### OS version

Window 11

### Printer

Klipper Ender 3

### How to reproduce

Visible edge on this object should no get any support. Support is set to normal manual, the support region is painted red and also additionally a support blocker is added on those faces. But still Orca produces support there. This area prints fine with just extra cooling. Support destroys the surface quality a lot there.

Geometry
[image]

Painting and blocker
[image]

It still adds support here:
[image]

### Actual results

Support is produced where it is blocked with painting and blocker

### Expected results

Do not add support in those areas.

### Project file & Debug log uploads

[support-blocker-bug-orca.zip](https://github.com/SoftFever/OrcaSlicer/files/13200204/support-blocker-bug-orca.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Noisyfox, 2023-11-10:
Use snag support and you'll get exactly what you want:
[image]

The grid support will by design expand outwards to some extends, which is not affected by support blocker.

AisonSu, 2023-12-08:
> Use snug support and you'll get exactly what you want: ![image](https://private-user-images.githubusercontent.com/1537155/282116160-b386e232-c6f8-4e98-a9d0-df76fe1d0281.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDIwMjMyNjksIm5iZiI6MTcwMjAyMjk2OSwicGF0aCI6Ii8xNTM3MTU1LzI4MjExNjE2MC1iMzg2ZTIzMi1jNmY4LTRlOTgtYTlkMC1kZjc2ZmUxZDAyODEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQUlXTkpZQVg0Q1NWRUg1M0ElMkYyMDIzMTIwOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzEyMDhUMDgwOTI5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZjA0MTk1MTg1ODg5Y2RlY2E0ODk1M2IyZDY2YjY0ZTUxNjA4MjM4YTBiMDg3YmIzMmQ4ZGNmYzFmN2QwZTBmNyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.r417Z-VNJz8BzFm8B2UdfSnQbXRBlBBIktg_zAN6bxU)
> 
> The grid support will by design expand outwards to some extends, which is not affected by support blocker. Snug support does not do this (hens the name).

Still no work

closed 2024-03-16 by github-actions[bot]

Gornerrrrr, 2024-09-02:
The support blocker still cant work in Orca2.1.1 : (

Dochartaigh, 2025-01-31:
> The support blocker still cant work in Orca2.1.1 : (

Still doesn't work in 2.2.0 on Mac... (and assume PC as well).