OrcaSlicer/OrcaSlicer #2547: Hogging RAM when using Tree supports

marcoa131, 2023-10-30:
### OrcaSlicer Version

1.8.0-beta1

### OS version

Windows 10

### Additional system information

CPU: Xeon E3-1271 v3
Memory: 32,0 GB DDR3
Graphics: Nvidia RTX2060; Driver Version 31.0.15.3623

### Printer

V0.2

### How to reproduce

Slice any file with painted supports and supports configs set to tree (auto) (because auto tree supports don't work) using "Default" Style.

### Actual results

[image]
After i press slice, the slicing gets stuck at 65% and the slicer imediatelly starts hogging RAM, as you can see in the image, and it takes hours to slice the file.

### Expected results

Should have sliced the file with supports after some seconds.

### Project file & Debug log uploads

[Grip.zip](https://github.com/SoftFever/OrcaSlicer/files/13209384/Grip.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Noisyfox, 2023-10-31:
This is expected. The default style has switched to organic tree which will use more memory.

marcoa131, 2023-10-31:
By more you mean up to 20 gigabytes? and take minutes to hours to slice a simple file? Seems unoptimized

foreachthing, 2023-10-31:
It only took a couple of seconds to slice and it barely touched to 600 MB RAM mark...
_Windows 11, i7-9700K, 32 GB RAM, RTX 2070_

Noisyfox, 2023-10-31:
> By more you mean up to 20 gigabytes? and take minutes to hours to slice a simple file? Seems unoptimized

Oops then there is a problem... I'll take a look.

marcoa131, 2023-10-31:
> It only took a couple of seconds to slice and it barely touched to 600 MB RAM mark... _Windows 11, i7-9700K, 32 GB RAM, RTX 2070_

but are you slicing using Organic default painted supports?

foreachthing, 2023-10-31:
I was using your settings in rhe 3mf..

closed 2024-02-21 by github-actions[bot]