OrcaSlicer/OrcaSlicer #2552: Internal solid infill can overlap itself 100% / overextruding

dartrax, 2023-10-31:
### OrcaSlicer Version

1.8.0 beta 1

### OS version

Win 10

### Printer

BL X1C

### How to reproduce

1. Open attached .3mf
2. Go to Layer 18

### Actual results

Two solid infill lines overlap 100% on the larger part. The larger part is just the smaller part extended by mirroring. Alternatively, set the Gap fill speed to any value (it's zero in the project file) and the lines will not overlap. I don't know why this has an influence.

[image]

### Expected results

No overlapping like on the smaller part

[image]

### Project file & Debug log uploads

[Overlap bug.zip](https://github.com/SoftFever/OrcaSlicer/files/13213798/Overlap.bug.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

SoftFever, 2023-10-31:
Hi 
This is an expected behavior. It's optimized path.

dartrax, 2023-11-02:
Thanks for the explanation. But when I set _Filter out tiny gaps_ to a value greater than 1 the whole line will be overlapped again. So this is not a fix right now...

Actually I like the Orca way much more, it is more intuitive and I'm able to define a threshold, no black or white. I was wondering if the value has a dimension, after experimenting with it it seems as if it's more a one-dimensional extrusion length (like in Simplify3D) and not a filling area or volume. May be add a "mm" dimension to the field?

SoftFever, 2023-11-02:
> Thanks for the explanation. But when I set _Filter out tiny gaps_ to a value greater than 1 the whole line will be overlapped again. So this is not a fix right now...

Acknowledged. It's indeed a bug.

> Actually I like the Orca way much more, it is more intuitive and I'm able to define a threshold, no black or white. I was wondering if the value has a unit, after experimenting with it it seems as if it's more a one-dimensional extrusion length (like in Simplify3D) and not a filling area or volume. May be add a "mm" unit to the field?

Yes, the threshold is defined in length(mm). Adding a unit is a good idea.

closed 2024-02-20 by github-actions[bot]