OrcaSlicer/OrcaSlicer #2557: AppImage does not launch on Linux Mint

deck55, 2023-10-31:
### OrcaSlicer Version

v1.7.0 and v1.8.0 beta1

### OS version

Linux Mint 21.2

### Printer

Ender 3 V2

### How to reproduce

Try to launch AppImagine

### Actual results

Nothing happens. Software is never launched

### Expected results

Software launches like it use to.

### Project file & Debug log uploads

Nothing to attach

### Checklist of files to include

- [ ] Log file
- [ ] Project file

brychanrobot, 2023-10-31:
I'm seeing the same thing on Fedora 38

This is the terminal log:
```
[2023-10-31 06:18:11.061959] [0x00007fc1049e3e80] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1
Unhandled unknown exception; terminating the application.
```

brychanrobot, 2023-10-31:
I tried running it with `./OrcaSlicer_Linux_V1.8.0-beta1.AppImage --datadir /tmp` and it launches just fine, so there must be some incompatibility with the old configs.

TexZeTech, 2023-10-31:
So what desktop environment are you using? 
- Cinnamon Edition
- MATE Edition
- Xfce Edition

or the Cinnamon "EDGE" Edition?

deck55, 2023-10-31:
Cinnamon is what I am running.

TexZeTech, 2023-10-31:
@deck55 So I just installed https://www.linuxmint.com/edition.php?id=305 into a Vm and updated it to current. I downloaded the .appimage and set it to executable and it started up with no problem. 

I would recommend going to /home/YOURUSERNAMEHERE/.config/OrcaSlicer/log and uploading whatever is in there.
You can also try running the .app in a terminal like this `./OrcaSlicer_Linux_V1.8.0-beta1.Appimage`

beyond that I have no ideas for you.

TexZeTech, 2023-11-01:
You could also try `journalctl -n 1500 > journalctldump.txt ` after trying to run the Appimage.

deck55, 2023-11-01:
Here is what is in the log file.

`2023-10-31 07:29:08.567943[Thread 0x00007faaa3b36e00]:Failed to parse configuration file "/home/cdeck/.config/OrcaSlicer/OrcaSlicer.conf": [json.exception.parse_error.101] parse error at line 1, column 1: syntax error while parsing value - unexpected end of input; expected '[', '{', or a literal
2023-10-31 07:29:08.577570[Thread 0x00007faaa3b36e00]:OrcaSlicer configuration file may be corrupted and is not abled to be parsed.Please delete the file and try again.

/home/cdeck/.config/OrcaSlicer/OrcaSlicer.conf

[json.exception.parse_error.101] parse error at line 1, column 1: syntax error while parsing value - unexpected end of input; expected '[', '{', or a literal`

I deleted the OrcaSlicer.conf file and it launched.

TexZeTech, 2023-11-01:
Don't forget to mark it solved then @deck55 . :smile:

closed 2023-11-01 by deck55