OrcaSlicer/OrcaSlicer #2565: A1 mini - G92 E0 error message

Eldenroot, 2023-10-31:
### OrcaSlicer Version

1.8.0b1

### OS version

Win11 64bit

### Printer

A1 mini

### How to reproduce

Add .stl file and you get the error 

[image]

### Actual results

[image]

### Expected results

No error message. Code is missing?

### Project file & Debug log uploads

No necessary

### Checklist of files to include

- [ ] Log file
- [ ] Project file

SoftFever, 2023-11-01:
It has been fixed in nightly build

closed 2023-11-01 by Eldenroot

VintarMfG, 2024-02-03:
[image]
[image]

And check the Z error ( 0.10), so as not to run into the table

(G4 ; )