OrcaSlicer/OrcaSlicer #2567: Vase mode for parts of the print

rkarlsba, 2023-11-01:
AFAICS vase mode can only be selected globally. It'd be very nice to allow for a figure to, say, have a solid base or at least a thicker one and then the top, as in a lamp shade, to use vase mode, perhaps even switching back on the top for a rim or similar. Would this be hard to implement?

HappyMark93, 2023-11-05:
It is possible, i have done it manually merging g-code solid on bottom, vase in mid and then again solid on top.
Transition layer is critical. Vase mode need to begin from zero level when z 0 offset is applied so that zero is on top surface of solid base.
Maybe it is possible to do right now with two solids stacked one on another, and vase mode selected for top model?

hungrierwtf, 2023-11-05:
This would be a really good feature. Or at least have it setable by height/layer, so that you could print the base with whatever settings you wanted, then transition to vase mode for the rest.

beatwinthewave, 2023-11-09:
+1 to have the vase mode for objects, good idea

Knaudler, 2023-12-17:
+1
we need Vasemode for parts or heigh range modifier - aktivate the button in the parts menu
every slicer can do this

ThinkBigPrintHuge, 2024-03-15:
+1
This is easily done in S3D. Not sure about other slicers as I have not tried. I am printing Rockets that are vase mode, but I need to add solid layers/infill to the tip to prevent sagging.

TuTanK5, 2024-06-10:
+1 would love to have this

closed 2024-09-16 by github-actions[bot]

Knaudler, 2024-09-16:
please add Vasemode at  heigh range modifier

soundfx37, 2026-01-30:
+1  Really useful feature for printing thick bases of vases.

Knaudler, 2026-10-16:
please reopen

rkarlsba, 2026-10-16:
+1 please reopen!