OrcaSlicer/OrcaSlicer #2574: Center of gravity toggle

liftbag, 2023-11-01:
**Is your feature request related to a problem? Please describe.**
PrusaSlicer has a toggle that allows you to view the object's center of gravity in print preview. Provided that the cog calculation is accurate, the tool, together with the modifiers, could be useful in some specific cases. I am thinking for example of floating or flying models, illusions etc.
Certainly nothing of vital importance, but it can be useful  so as not to waste on trial and error where the center of gravity of the printed object is relevant.

**Which printers will be beneficial to this feature**
N.A.

**Describe the solution you'd like**
Porting from PrusaSlicer? As long as there are no problems of the type that porting of the ruler is causing.

**Additional context**

COG of the printed object
[image]

Moved forward via localized infill modifier
[image]

Object that cannot remain erect with any uniform infill option

[image]

By modifying the infill in a localized way I can safely print the result.
[image]
[image]

liftbag, 2024-02-20:
The request is always present

ianalexis, 2026-11-03:
This feature was included and expanded in #16291 so its already included in 2.5.0

closed 2026-11-03 by ianalexis