OrcaSlicer/OrcaSlicer #2576: Is there any option to avoid supports in vertical holes?

oleksii-suprun, 2023-11-01:
I have a small model with vertical holes. The slicer generates supports inside holes that are 2.5mm in diameter. Unfortunately, it's impossible to remove supports from those holes. I'm looking for any way to prevent supports in holes.
[image]
[image]
[image]

igiannakas, 2023-11-01:
Orient the part side ways - it won't print without supports. Maybe organic can squeeze in there but its unlikely

oleksii-suprun, 2023-11-01:
@igiannakas thanks for the recommendation. I understand that supports are required and I have to use this for this model, I'm just curious if there is any way to avoid support inside holes. They are useless and do not support anything. In this case, Cura does not add supports inside holes for the same model.
[image]
[image]

igiannakas, 2023-11-01:
Yeap. You can add a support blocker in the lower layers and only out support between the two parts. However I would print it oriented on the flat side where the two holes are located. With the cylinder thing on the top side.

oleksii-suprun, 2023-11-01:
I've tried to block supports at the bottom, but that did not help.
[image]
[image]

Yep, I also tried to print it oriented as you described but had issues with adhesion. The model unsticks from the bed due to the small contact area (maybe I should try to use a brim, but I did not want to waste time on tries and failures)

Nevertheless, finally, I printed that model using Cura. Even with the strange rings that were sliced to be printed in the air, that issue did not cause any problems during the printing and I've already successfully printed two copies. But I'm a bit frustrated that I could not get that result in Orca Slicer.

Yen-Lung-Huang, 2024-03-15:
Try this: [Optimize your Supports with Support Blockers in Orca Slicer](https://www.youtube.com/watch?v=2uP7hVE3spw)