OrcaSlicer/OrcaSlicer #2578: It has a crashing bug when using calibration

darkck, 2023-11-02:
### OrcaSlicer Version

1.8.0 beta-1

### OS version

windows 11

### Printer

P1S

### How to reproduce

When calibration， it always crash

### Actual results

crash

### Expected results

cant use calibration in orca

### Project file & Debug log uploads

s

### Checklist of files to include

- [ ] Log file
- [X] Project file

SoftFever, 2023-11-04:
Hi @darkck 

Please provide more details.

mgaruccio, 2024-01-31:
I think I may be encountering the same problem but on macos.  when I attempt to run the dynamic flow calibration using the pattern option orca immediately crashes to desktop. I've tried with both system preset and user-defined filaments and different start/stop points for the K value with no luck.  switching to the "line" calibration method allows the calibration to actually continue.  I'm also using a P1S and so using manual calibration.

the problem seems to be a a type error (log snippet below) but I'm including the entire log in case it's helpful.  The entries at and around 2024-01-31 09:43:31 are the ones that should be relevant.

let me know if there's other info I can provide to be helpful here.

```
2024-01-31 09:43:31.621221[Thread 0x0000000170a8b000]:process_task{ type:RemoveBackup, id:53, path:/var/folders/0f/yr7rcvz55yb9fp77rx2l08xc0000gn/T/orcaslicer_model/Wed_Jan_31/09_43_31#54044#53, object:0, extra:1} and interval = 10
2024-01-31 09:43:31.621467[Thread 0x00000001ff109e00]:09:43:31: Error: OrcaSlicer got an unhandled exception: ConfigOptionFloatOrPercent: Assigning an incompatible type
2024-01-31 09:43:31.621470[Thread 0x00000001ff109e00]:Uncaught exception: ConfigOptionFloatOrPercent: Assigning an incompatible type
2024-01-31 09:43:31.661622[Thread 0x00000001ff109e00]:~NetworkAgent, this 0x60000074a800, network_agent=0x60000074a810, destroy_agent_ptr=1, ret 0
2024-01-31 09:43:31.683209[Thread 0x00000001ff109e00]:~WebViewPanel Start
2024-01-31 09:43:31.683249[Thread 0x00000001ff109e00]:~WebViewPanel End
2024-01-31 09:43:31.693974[Thread 0x00000001ff109e00]:ImageGrid::SetStatus: No printers.
2024-01-31 09:43:31.694646[Thread 0x00000001ff109e00]:~PrinterWebView Start
2024-01-31 09:43:31.694649[Thread 0x00000001ff109e00]:~PrinterWebView End
2024-01-31 09:43:31.716712[Thread 0x00000001ff109e00]:stop, enter
2024-01-31 09:43:31.716843[Thread 0x00000001ff109e00]:update_slice_result_valid_state: plate 0 , update slice result from 0 to 0
2024-01-31 09:43:31.716854[Thread 0x00000001ff109e00]:update_slice_result_valid_state: plate 0 , update slice result from 0 to 0
2024-01-31 09:43:31.716907[Thread 0x00000001ff109e00]:update_slice_result_valid_state: plate 1 , update slice result from 0 to 0
2024-01-31 09:43:31.717565[Thread 0x00000001ff109e00]:update_slice_result_valid_state: plate 1 , update slice result from 0 to 0
2024-01-31 09:43:31.717651[Thread 0x00000001ff109e00]:remove_backup /var/folders/0f/yr7rcvz55yb9fp77rx2l08xc0000gn/T/orcaslicer_model/Wed_Jan_31/09_41_10#54044#49, true
2024-01-31 09:43:31.717711[Thread 0x00000001ff109e00]:process_ui_task{ type:RemoveBackup, id:53, path:/var/folders/0f/yr7rcvz55yb9fp77rx2l08xc0000gn/T/orcaslicer_model/Wed_Jan_31/09_43_31#54044#53, object:0, extra:1} and interval = 10
2024-01-31 09:43:31.717724[Thread 0x0000000170a8b000]:process_task{ type:RemoveBackup, id:7, path:/var/folders/0f/yr7rcvz55yb9fp77rx2l08xc0000gn/T/orcaslicer_model/Wed_Jan_31/09_41_10#54044#49, object:0, extra:1} and interval = 10
2024-01-31 09:43:31.718088[Thread 0x00000001ff109e00]:process_ui_task: remove all of backup path /var/folders/0f/yr7rcvz55yb9fp77rx2l08xc0000gn/T/orcaslicer_model/Wed_Jan_31/09_43_31#54044#53
2024-01-31 09:43:31.726070[Thread 0x00000001ff109e00]:reset: current result id 0 
2024-01-31 09:43:31.726530[Thread 0x00000001ff109e00]:reset: current result id 4294967295 
2024-01-31 09:43:31.726580[Thread 0x00000001ff109e00]:reset: current result id 0 
2024-01-31 09:43:31.754947[Thread 0x00000001ff109e00]:~GUI_App: enter
2024-01-31 09:43:31.754960[Thread 0x00000001ff109e00]:~GUI_App: destroy app_config
2024-01-31 09:43:31.754973[Thread 0x00000001ff109e00]:~GUI_App: destroy preset_bundle
2024-01-31 09:43:31.758693[Thread 0x00000001ff109e00]:~GUI_App: destroy preset updater
2024-01-31 09:43:31.758713[Thread 0x00000001ff109e00]:~GUI_App: exit
```
[debug_Wed_Jan_31_09_41_09_54044.log](https://github.com/SoftFever/OrcaSlicer/files/14113191/debug_Wed_Jan_31_09_41_09_54044.log)

closed 2024-05-09 by github-actions[bot]