OrcaSlicer/OrcaSlicer #2581: PA: Plate boundaries error

dJOS1475, 2023-11-02:
### OrcaSlicer Version

1.8.0

### OS version

macOS 14.1

### Additional system information

"2018" Intel Mac Mini

### Printer

Custom w/ 225x225mm bed and -10mm X & Y origin

### How to reproduce

Create PA advance Line test eg 0.02 - 0.08 with 0.002 increments
Slice model 

### Actual results

Error: A G-code path goes beyond plate boundaries

### Expected results

No error. 
The sliced model is clearly within bounds

[image]
[image]

### Project file & Debug log uploads

[debug_Thu_Nov_02_13_35_13_35539.log.0.zip](https://github.com/SoftFever/OrcaSlicer/files/13234060/debug_Thu_Nov_02_13_35_13_35539.log.0.zip)

[Pressure Advance Test.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/13234072/Pressure.Advance.Test.3mf.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

A-Printer, 2023-11-10:
I have a similar problem with PA calibration, while generating PA Line with default options. In my case, the lines are generated partly out of the build plate.

**OrcaSlicer Version**

1.8.0-beta2

**OS version**

Same issue on macOS and Windows

**Printer**

FlashForge Adventurer 5M

**How to reproduce**

Create PA advance Line with default settings
Slice model

**Actual results**

The "PA Test" texts appears centered on the build plate. But when I slice it, the lines appear partially out of the plate (see screenshot).

[image]

If I move the "PA Test" text on the plate and slice I get this very unhelpful error message:

[image]

A-Printer, 2023-11-13:
If I set the origin to 0, 0 instead of the default 110, 110 (which makes the arrows appear at the center of the plate instead of the lower left) the preview is now shown correctly inside the build plate boundaries, but when I give the gcode file to the printer it detects that it's going to print outside of boundaries. I observed the same issue with the Max flowrate calibration.

closed 2024-02-20 by github-actions[bot]

H8ted666, 2024-12-27:
It's still doing it. For whatever reason it sets the center to 0x0 regardless of settings.

wmahmoud, 2026-05-08:
This is still happening as of today in 2.3.2.

OrcaSlicer Version

2.3.2.

OS version

Windows 10

Printer

Elegoo Centauri Carbon

How to reproduce

Create PA advance Line with default settings
Slice model