OrcaSlicer/OrcaSlicer #2582: Skipping issues is not supported in latest 1.8.0-beta1 version

Gieryic, 2023-11-02:
### OrcaSlicer Version

1.8.0-beta1

### OS version

Windows 11

### Printer

Bambu Lab X1C

### How to reproduce

Prior versions of Orca Slicer would allow for skipping objects directly on the X1C printer screen (mid-print), but this latest version throws an error saying "skipping objects is not supported on slicer versions prior to Bambu 1.6".

### Actual results

Can not skip individual objects (e.g. in case of failed printing)

### Expected results

Displays error saying "skipping objects is not supported on slicer versions prior to Bambu 1.6".

### Project file & Debug log uploads

[Pokeball Closed.zip](https://github.com/SoftFever/OrcaSlicer/files/13234812/Pokeball.Closed.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

closed 2024-02-20 by github-actions[bot]

Danitegue, 2024-03-23:
This issue should not be closed...  
As an example:

Any 2 colors piece, where the lower half is one color, and the upper half is another color. 
To print mutiple instances of that piece, I clone it, and then slice, using Orca 2.0.0-beta.
Then I send the order to print it into my X1C.

While printing, if I want to skip one of the instances (like when a spagetti happens in only one of the instances to be printed, but all the others are still ok), the screen selector does not allow me to skip the problematic instance, it instead propose me to skip all  the bottom halfs of all the instances, or all the upper halfs of all the instances... absurd...

This does not happen when slicing and printing with the Bambulab studio software (v1.8.4.51)

ImSadPanda, 2024-04-29:
The same thing is happening for me with both my P1S and P1P printers

nnobuddy, 2024-11-20:
I'm having the same problem. My printer says:
Skipping objects is not supported in:
-Over 64 objects in single plate
-Bambu Studio slicer lower than
V1.6
-Perform other non-printing tasks
-Only one object in printing

My Bambu Studio is 1.10.1.50.

Due to **extremely poor performance** of my new X1C (every other print fails, extreme stringing, and many more...) this bug should be fixed ASAP. I'm afraid I will have to return the printer to the dealer. It's a big disappointment.

P.S. Cannot paste image here. :-(