OrcaSlicer/OrcaSlicer #2587: Minimum Z-hop between all travel moves?

scottmudge, 2023-11-02:
Forgive me if this is already a feature, but I haven't been able to find it.

I've set the z-hop height in the filament settings, but this only gets applied to post-retraction travels. Frequently, however, retraction does not occur between travel moves when extruding infill, for example. This causes my hotend to "rattle"/scrape over the infill pattern, which sounds pretty horrible.

If there is no feature to add a minimum z-hop for *all* travel moves, can it be added?

Thanks!

igiannakas, 2023-11-02:
Yes, disable this option (reduce infill retraction). Then it will retract and zhop across all moves greater than your travel distance threshold (printer settings -> extruder).

[image]

scottmudge, 2023-11-02:
Ah so that's where that setting was hiding. Thanks!

I may add a feature to just so a z-hop within infill and not retract, to save on time not needing to do a z move. But for now this will work.

closed 2023-11-02 by scottmudge

igiannakas, 2023-11-02:
> Ah so that's where that setting was hiding. Thanks!
> 
> I may add a feature to just do a z-hop within infill and not retract first, to save time.

If you zhop without retraction you'll get plenty of stringing, which even in infill I'd consider as undesirable personally.