OrcaSlicer/OrcaSlicer #2588: Internal bridge uses wrong filament

dartrax, 2023-11-02:
### OrcaSlicer Version

1.8.0 beta 1

### OS version

Win 10

### Printer

BL X1C

### How to reproduce

1. Open attached .3mf project
2. Slice, go to Layer 5
3. Observe the internal bridge in the orange rect has the wrong filament, should be orange, too
4. Now cut the part like shown in the second image
5. Slice again
6. Observe the internal bridge in the orange rect has the right filament

### Actual results

[image]

### Expected results

[image]

### Project file & Debug log uploads

[Color Bug.zip](https://github.com/SoftFever/OrcaSlicer/files/13238626/Color.Bug.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

dartrax, 2024-02-12:
This bug is still present in 2.0.0-dev

dartrax, 2024-05-13:
This bug is still present in 2.0.0 release.

dartrax, 2024-08-12:
*unstale

dartrax, 2024-11-11:
*unstale

ndevenish, 2025-01-04:
I have also just run into this issue trying to print https://www.printables.com/model/104390-wingspan-card-box-w-magnetic-lid. It isn't present in Bambu Studio.

I might try to bisect it, although last time I identified the commit causing a bug it was closed as "not planned" by the bot, so not really sure it is worth the work.

dartrax, 2025-04-05:
*unstale

dartrax, 2025-10-07:
*unstale

Felix14-v2, 2026-10-06:
Can confirm in 2.5.0-nightly (eb5b9a77), the modifier doesn't change the color of bridges.
[image]

Your file crashes the app for some reason, so here's the fixed one: [Color Bug (2.5.0).zip](https://github.com/user-attachments/files/33082854/Color.Bug.2.5.0.zip)