OrcaSlicer/OrcaSlicer #2590: Sync with latest Bambu Studio GitHub code

Eldenroot, 2023-11-02:
**Is your feature request related to a problem? Please describe.**
2 days ago BL pushed new code on their github 

**Which printers will be beneficial to this 
All

**Describe the solution you'd like**
Sync what is useful

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Ocraftyone, 2023-11-04:
Is there a specific feature you are looking to  have ported?

Eldenroot, 2023-11-04:
Cut tool

Eldenroot, 2023-11-04:
And you know, always less bugs and more handy features are welcomed

Eldenroot, 2023-11-10:
Now Bambu studio version 1.8 (beta) is out :)

Eldenroot, 2023-12-02:
Done in 1.9 (at least all what was released by BL studio on Github until this date).

closed 2023-12-02 by Eldenroot