OrcaSlicer/OrcaSlicer #2591: When a Z-Offset is set, klipper returns "Out of range" error

SejeAnd, 2023-11-02:
### OrcaSlicer Version

1.8.0

### OS version

Windows 11

### Printer

Anycubic Vyper

### How to reproduce

Set a Z-Ofsset in the printer options.
Run it on a klipper machine, and it will after the start g-code return an "Out of range" error.

### Actual results

As mentioned, klippe will after the start g-code return an "Out of range" error.
[image]

### Expected results

It shouldnt be an issue. It should just start printing.

### Project file & Debug log uploads

Klipper log: 
[klippy.log](https://github.com/SoftFever/OrcaSlicer/files/13243988/klippy.log)
Moonraker log: 
[moonraker.log](https://github.com/SoftFever/OrcaSlicer/files/13243991/moonraker.log)
G-code: 
[FirstLayerCalibration-FirstLayer_PLA_7m8s.txt](https://github.com/SoftFever/OrcaSlicer/files/13244020/FirstLayerCalibration-FirstLayer_PLA_7m8s.txt)

### Checklist of files to include

- [X] Log file
- [X] Project file

SoftFever, 2023-11-03:
You will need to adjust Stepper Z's position_min.

SejeAnd, 2023-11-03:
> You will need to adjust Stepper Z's position_min in Klipper's config file

Even tho it's set in the slicer? If so, where should i add the variable in the config?

SoftFever, 2023-11-03:
> > You will need to adjust Stepper Z's position_min in Klipper's config file
> 
> Even tho it's set in the slicer? If so, where should i add the variable in the config?

It has nothing to do with the slicer. 
Basically you have klipper configed to say the minimun z can't be lower than e.g. -0.5 mm, but you configure the slicer to offset the z so that it yeilds -0.6mm values.
You will have to changed it in klipper.cfg

Anyway, I suggest you ask for help in Klipper forurm or discord.
As here we only discuss OrcaSlicer bugs.

closed 2023-11-05 by SoftFever