OrcaSlicer/OrcaSlicer #2592: Dynamic Flow Calibration -- Crashes when attempting Manual Calibration

arhdrums, 2023-11-02:
### OrcaSlicer Version

1.8.0-beta 1

### OS version

Windows 11

### Printer

Bambu Lab X1 Carbon

### How to reproduce

1. Click on Calibration Panel (Bambu Lab)
2. Click on Manual Calibration
3. Attempt to start a calibration with either the Line or Pattern methods.

### Actual results

An error message appears, then OrcaSlicer closes.

"Unhandled unknown exception; terminating the application"

[image]

### Expected results

A manual calibration should be successfully generated and sent to the printer.

### Project file & Debug log uploads

[debug_Thu_Nov_02_16_21_03_16500.log.zip](https://github.com/SoftFever/OrcaSlicer/files/13244233/debug_Thu_Nov_02_16_21_03_16500.log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

SoftFever, 2023-11-05:
Fixed in 1.8 beta 2

closed 2023-11-05 by SoftFever

arhdrums, 2023-11-05:
Thank you! I can confirm that it is now working for me.

Thanks for quickly fixing the issue, and take care!

arhdrums, 2023-11-05:
> Fixed in 1.8 beta 2

Thank you! I can confirm that Beta 2 has fixed the issue for me.

Thanks again and take care!

kurohouou, 2023-12-06:
Running 1.8.1 and getting this error now.

Belliqueu, 2023-12-14:
Also getting this error in 1.8.1 Windows 10

jintaogao, 2023-12-23:
happening in 1.8.1 Mac OS 14.1

ronx, 2024-06-28:
This is happening with Pattern calibration in 2.1.0 and 2.1.1. Line works fine.

Freekers, 2024-06-28:
> This is happening with Pattern calibration in 2.1.0 and 2.1.1. Line works fine.

Can confirm, same issue here.

TomN7, 2024-06-28:
Also getting a crash on every attempt to use the pattern calibration in 2.1.0 and 2.1.1. 
Generating a pattern manually with the calibration option in the toolbar works, but not the step-by-step calibration tab.

rmzalbar, 2024-07-17:
> Also getting a crash on every attempt to use the pattern calibration in 2.1.0 and 2.1.1. Generating a pattern manually with the calibration option in the toolbar works, but not the step-by-step calibration tab.

Me too, v. 2.1.1. Settings: Pattern, 0.022 to 0.032 step 0.001. Works in current Bambu Studio.