OrcaSlicer/OrcaSlicer #2593: Port "printing_by_object_gcode"  feature from BambuStudio

IntenseWiggling, 2023-11-02:
**Is your feature request related to a problem? Please describe.**
Inability to insert custom gcode in-between objects when using the 'By object' print sequence.

**Which printers will be beneficial to this feature**
Any printer.

**Describe the solution you'd like**
Port the functionality from BambuSlicer, which was committed a few days ago:

Ref: https://github.com/bambulab/BambuStudio/commit/5076a3008ab085d4196ee9ef1be2b2b2766911a7

**Describe alternatives you've considered**
Port from PrusaSlicer?

**Additional context**
Would allow for the ability to do things like performing nozzle / brush wipes between objects, or adjust settings between objects not available in by-boject settings (e.g. adjust the Z offset , retraction (firmware) settings), or send commands to klipper.

Ocraftyone, 2023-11-04:
I'll take a look at it. Seems like the feature is mostly there, just commented out.

IntenseWiggling, 2024-01-15:
Noticed with the 1.8 beta commit that the `Printing by object G-Code` is available in the UI. Though, nothing that I can see in the actual release notes?

https://github.com/SoftFever/OrcaSlicer/commit/d8dd8fa63462f7b5545dce1491125dbc55ded63c

But it seems to have not actual impact. The generated gcode doesn't have the input of that field inserted between objects.

Ocraftyone, 2024-01-15:
It is disabled still. I have not yet determined if it works when in commented. Give me a moment and I'll create a PR with it uncommented

Ocraftyone, 2024-01-15:
Please try the build from the PR mentioned above.

PR #3691 (2024-01-21): Re-enable printing by object gcode
BBL has printing by object Gcode disabled. This just re enables it. Fixes #2593
closed 2024-01-21 by SoftFever