OrcaSlicer/OrcaSlicer #2599: Filament chooser shows only my initial imported filaments. Impossible to add new.

arittner, 2023-11-03:
### OrcaSlicer Version

1.7.0

### OS version

Windows 11 64bit

### Printer

Creality Ender 3 S1 pro with Klipper

### How to reproduce

Add Filament with the Combo box:

[image]

Choose all:

[image]

Click on Confirm:

Nothing added:

[image]

I also tried it with the small gear near the "Filament" section. Nothing happened.

I also tried to rerun the assistant via help menu: Nothing happened.

It's not possible to load filament setting from the presets. I can only add custom filament settings, by changing existing and saving the settings under a new name.

In the log file I see a huge list of filament settings (as JSON file), but after the list:

```
2023-11-03 15:40:43.098343[Thread 0x00001848]:Slic3r::GUI::GuideFrame::SaveProfilevendor_name Custom, model_name Generic Klipper Printer, nozzle 0.4 selected
2023-11-03 15:40:43.099342[Thread 0x00001848]:Slic3r::GUI::GuideFrame::SaveProfilevendor_name Creality, model_name Creality Ender-3 S1 Pro, nozzle 0.4 selected
2023-11-03 15:40:43.117852[Thread 0x00001848]:GuideFrame returned ok
2023-11-03 15:40:43.117852[Thread 0x00001848]:No bundles need to be installed from resource directory
2023-11-03 15:40:43.117852[Thread 0x00001848]:No bundles need to be removed
2023-11-03 15:40:43.285144[Thread 0x00001848]:GuideFrame applied
```

### Actual results

On any steps the filament list is not changing. I cannot add filament settings.

### Expected results

The filament settings should appear.

### Project file & Debug log uploads

[debug_Fri_Nov_03_15_13_56_15880.log.zip](https://github.com/SoftFever/OrcaSlicer/files/13251527/debug_Fri_Nov_03_15_13_56_15880.log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

arittner, 2023-11-03:
I tried to add a new TPU type manually, but Orca Slicer deletes everything on startup.

I'm completely unsure, what going wrong:

Here are my steps:

Added in `./system/Creality.json` (in `filament_list`):

```json
		{
			"name": "fdm_filament_tpu",
			"sub_path": "filament/fdm_filament_tpu.json"
		},
		{
			"name": "Creality Generic TPU",
			"sub_path": "filament/Creality Generic TPU.json"
		}
```

Because I use `Creality Ender-3 S1 Pro 0.4 nozzle` and the printer model is `Creality Ender-3 S1 Pro`, I changed `./system/Creality/machine/Creality Ender-3 S1 Pro.json` and added `Creality Generic TPU` in the default filament list:

```json
    "default_materials": "Creality Generic PLA;Creality Generic PETG;Creality Generic ABS;Creality Generic TPU"
```

Then I created two files in the `./system/Creality/filament/` folder:

`./system/Creality/filament/Creality Generic TPU.json` with `compatible_printers` that includes `"Creality Ender-3 S1 Pro 0.4 nozzle"`:

```json
{
    "type": "filament",
    "filament_id": "GFU99",
    "setting_id": "GFSA04",
    "name": "Creality Generic TPU",
    "from": "system",
    "instantiation": "true",
    "inherits": "fdm_filament_tpu",
    "filament_max_volumetric_speed": [
        "3.2"
    ],
    "compatible_printers": [
		"Creality CR-10 V2 0.4 nozzle",
		"Creality CR-10 Max 0.4 nozzle",
		"Creality Ender-3 0.4 nozzle",
		"Creality Ender-3 V2 0.4 nozzle",
		"Creality Ender-3 S1 0.4 nozzle",
		"Creality Ender-3 S1 Pro 0.4 nozzle",
		"Creality Ender-5 0.4 nozzle",
		"Creality Ender-5 Plus 0.4 nozzle",
		"Creality Ender-5 Pro (2019) 0.4 nozzle",
		"Creality Ender-5S 0.4 nozzle",
		"Creality Ender-5 S1 0.4 nozzle",
		"Creality Ender-6 0.4 nozzle",
		"Creality K1 (0.4 nozzle)",
		"Creality K1 (0.6 nozzle)",
		"Creality K1 (0.8 nozzle)",
		"Creality K1 Max (0.4 nozzle)",
		"Creality K1 Max (0.6 nozzle)",
		"Creality K1 Max (0.8 nozzle)",
		"Creality Ender-3 S1 Pro 0.4 nozzle - Klipper"
    ]
}
```

Finally, I created the file `./system/Creality/filament/fdm_filament_tpu` (which is a plain copy from the `custom/filament` folder).

If I start Orca Slicer with this modified configuration, the application deletes all the changes, and I'm back with the old settings.

Maybe this is the wrong approach. On this way, I cannot add TPU to my settings - because Orca overwrites the system settings. But why? Am I missing some necessary changes, or is Orca replacing the data from another source?

This means I must create from scratch a plain TPU filament material setting. I cannot use a setting from the assistant, and I cannot import filament settings.

arittner, 2023-11-03:
Ok, I made the changes in the application resources folder.

My first try was not successful (I thought at first). But I started the filament chooser, and now I see (unchecked) the `Creality Generic TPU`. I checked the filament type, and now it's added to my combo box (I guess it's an entry in `OrcaSlicer.conf`)

I do understand that there is a fixed relationship between compatible filament types and printers. However, it is absolutely unclear to me why there is no way to add your own filament types based on generic filaments. It is rather the common rule that you use all kinds of materials from all kinds of manufacturers on one printer. A static relationship is helpful, but it should be extendable.

What I can't understand is that Orca overwrites the `AppData` with the data from the application resource folder. That simply must not happen. If you change these properties as a user, an application must not overwrite the information. This is completely atypical for applications, especially under Windows. 

Consequently, this means that Orca will delete my config with the next update. 

I know, in `AppData` is a `user/default` folder, but here I cannot create derived filament materials and mix the information from a generic printer for my Creality printer. Orca entirely ignores a filament inherited from a `fdm_filament_*.json` because the file is not in the `system/Creality/filament` folder, and Orca also ignores `fdm_filament_*.json` files in `user/default/filament` (at least I haven't found out how it works).

I hope, it is possible to give a user a better idea, how the filament relationships work. It's hard to understand that I'm able to see many filaments for different printers, but I cannot use the selected generic filaments for a specific printer. Besides this user experience, overwriting the AppData is IMHO a bug. Additionally, it's a huge flaw, that a user cannot re-use filament types from other printers (e.g. from the generic printer). All these issues in combination make Orca very static in the way to handle materials.

lowlifelowrider, 2024-02-15:
same issue on my k1. find a fix?

BertranK, 2024-03-08:
I have the same issue in Ver1.9.1 with my K1 Max,

BertranK, 2024-03-15:
I fixed it via reinstall Orca Slicer with deleting every Orca Slicer folders under AppData\Roaming&Local

closed 2024-06-22 by github-actions[bot]

sharky98, 2024-08-13:
Not stale. Original post and comment still need clarification.