OrcaSlicer/OrcaSlicer #2600: "no cooling on first layer" sets fan to 100

Brace91, 2023-11-03:
### OrcaSlicer Version

1.8.0-beta1

### OS version

macOS 14.0

### Printer

Bambu X1

### How to reproduce

1, in filiment profile set no cooling in first layer to x.
2, in preview you will see that the first x layers fan speed is 100

### Actual results

First layer fan speed is 100

### Expected results

First layer fan speed should be 0

### Project file & Debug log uploads

[Air Boat Orca 2.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/13252401/Air.Boat.Orca.2.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

SoftFever, 2023-11-04:
Can you provide more details? It works properly as what I can see
[image]

Brace91, 2023-11-05:
Plate 15 worked as expected but here’s a screenshot of plate 1 with the issue. 

￼

dartrax, 2023-11-22:
I have the same problem and I found that it happens when "Fan speed-up time" is set in a Multicolor print.
This bug is present in 1.8.0 release.
Here is an example.
[image]

[Fan speed bug.zip](https://github.com/SoftFever/OrcaSlicer/files/13444192/Fan.speed.bug.zip)

dartrax, 2024-02-21:
Still present in 1.9.0 release

cubitik, 2024-03-03:
> Still present in 1.9.0 release

Update: Still present in 1.9.1 release. Doesn't show a full fan speed in preview but print starts with full fan speed.

dartrax, 2024-06-02:
Still present in 2.1.0-beta

dartrax, 2024-09-01:
unstale

real-artswan, 2024-10-13:
2.2.0-beta confirming

audemaeva, 2024-11-02:
Still present in 2.2.0
I set my temp this way : the fan is always at 100% when the print starts
[image]

fabio-s-franco, 2025-01-06:
> 2.2.0-beta confirming, setting fan speed-up and kick-start to 0 solves it

I believe this may not work for everyone due to timing. In my case if I execute this on the console:

```
M106 P3 255
UPDATE_DELAYED_GCODE ID=part_fan_off DURATION=2
```

The fan does not stop.

If instead do this:

```
M106 P3 255
UPDATE_DELAYED_GCODE ID=part_fan_off DURATION=3
```

The fan reaches full speed for short moment and then turns off.
This is the macro for the gcode if anyone wants to test (Delayed gcode must be enabled in klipper):
```
[delayed_gcode part_fan_off]
gcode:
    M106 S0
```

And finally, my workaround is to use this on **before** layer change (Machine G-code in OrcaSlicer) and checking whether it is about to start the first layer:
```
{if layer_num == 0}
  UPDATE_DELAYED_GCODE ID=part_fan_off DURATION=3
{endif}
```

closed 2025-04-06 by Brace91