OrcaSlicer/OrcaSlicer #2602: missing option "ensure vertical shell thickness" partly breaks single-wall slicing?!

AmaZz0r, 2023-11-03:
When slicing models of parts for rc-planes that are designed to be printed without infill (single wall, printed ribs, close to vase mode), there is internal solid infill added despite setting infill to 0.

I also tried basically every other setting that might have something to do with it, but couldnt get rid of the internal solid infill.

This does not only add weight, but also causes print defects when printing with active foaming (lightweight) material. Why: Retractions are compensated with "extra length on restart", but the internal solid infill has a very close spacing; therefore, excess material accumulates.

This wasnt a problem when the option "ensure vertical shell thickness" was still present. I think it disabled internal solid infill generation.

Using vase mode is not an option, as these designs sometimes encorporate top+bottom layers and inevitable travel  moves.

## pay attention to the additional purple lines in the OrcaSlicer preview

How it looks in OrcaSlicer:
[image]

How it should look:
[image]

real life defects with LW-filament:
[image]

Noisyfox, 2023-11-04:
Can you share the project file?

AmaZz0r, 2023-11-04:
> Can you share the project file?

Sure, thanks for looking into this.

[WingPiece_sample.zip](https://github.com/SoftFever/OrcaSlicer/files/13256647/WingPiece_sample.zip)

AmaZz0r, 2023-11-04:
The sliced file just reminded me of the floating bridges that are added too.

They're not necessary as well. Don't know if they'd cause any problems.

But normally, they don't belong there.

The shared project has one at layer 216.

It seems they are generated to support ribs that would otherwise start printing in the air.

Maybe theyre actually helpful, I dont know.

Otherwise, using bridge speed and bridge cooling, the ribs would probably print out fine without a whole bridged surface beneath.

Noisyfox, 2023-11-04:
Can you also send me the original model/project as well? If that concerns you, you can email me.

AmaZz0r, 2023-11-04:
what do you mean by original model/project?

Noisyfox, 2023-11-04:
The project before slicing. The 3mf file you uploaded earlier contains only the resulting gcode which make it less useful because I could not re-slicing it with different parameter/version.

AmaZz0r, 2023-11-04:
ah, sorry. thought that the settings etc. were included in the .3mf after slicing as well.

[WingPiece_sample_project.zip](https://github.com/SoftFever/OrcaSlicer/files/13256841/WingPiece_sample_project.zip)

SoftFever, 2023-11-04:
looks like a setting issues.
Can you check if this is expected?
[image]

[WingPiece_sample_project_corrected.zip](https://github.com/SoftFever/OrcaSlicer/files/13256992/WingPiece_sample_project_corrected.zip)

AmaZz0r, 2023-11-04:
> looks like a setting issues. Can you check if this is expected? [image]
> 
> [WingPiece_sample_project_corrected.zip](https://github.com/SoftFever/OrcaSlicer/files/13256992/WingPiece_sample_project_corrected.zip)

nope. that's not how it should look like. it's missing all of the internal ribs that way.

SoftFever, 2023-11-04:
I see.
Seems we still need expose the option for user to turn off it.
@Noisyfox what do you think?

AmaZz0r, 2023-11-04:
[image]

that's the result you'd normally want.

Noisyfox, 2023-11-05:
Haven't figured out better solution for this issue, so I think I will add the option back.

closed 2023-11-06 by SoftFever

Annuit20, 2023-11-07:
Can we add an option for a tolerance at which point vertical shell thickness kick in?
At minor overhangs most of the time there is not need we can compensate with more walls if we want.
But some geometries would benefit from these extra lines.

PR #2626 (2023-11-06): Add back the option to disable `ensure vertical shell thickness` and `detect narrow internal solid infill`
Partly reverted commit 075a08bca8a033c896c48f2d3612f74ff7a19716
This should fix #2602 and fix #2623