OrcaSlicer/OrcaSlicer #2604: THANK YOU for OrcaSlicer!

discip, 2023-11-04:
# Just needed to thank everyone who made OrcaSlicer what it is today. 😃 👍🏻 🚀 

Especially:
@SoftFever
@Noisyfox
@igiannakas
@Ocraftyone

SoftFever, 2023-11-04:
Thank you for the kind words!
I want to express my appreciation to you as well.
Thank you for actively testing and providing feedback for the PR/Dev builds.

discip, 2024-02-12:
@github-actions 😄
In my opinion this should be updated every time a new feature is implemented!
Not only do I use this great software myself, but I also recommend it to anyone who asks which slicer to use.

# Many thanks in particular to the people mentioned in the first post for persistently maintaining OrcaSlicer and adding all of those great handy features.
(Not tagging them to keep annoyance-rates low.) 😊

closed 2024-05-21 by github-actions[bot]