OrcaSlicer/OrcaSlicer #2606: Infill Color Incorrect

mydevpeeps, 2023-11-04:
### OrcaSlicer Version

1.7.0

### OS version

Windows 10

### Additional system information

N/A

### Printer

Bambu Lab X1C w/AMS

### How to reproduce

1. Import STL (in this case a jar lid)
2. Set two colors in filament selection. First black, second yellow.
[image]
3. Place lid so top is flat with the plate.
4. Select lid object and choose to paint colors on it.
[image]
5. Select yellow and height and only select the part of the lid that is the edges and not the top.
[image]
6. Slice the plate and be in Preview tab.
[image]
7. Pull top layer slider down so you can see the infill. 
[image]

NOTE: I tried to switch the colors around in the filament list and repaint it, but it made no difference.
[image]

### Actual results

The color of the infill is the second color chosen.

### Expected results

The color of the infill matches the color of the edge OR first color if that edge was not painted.

Feature: It would be great if we could choose/override the infill color like we can Supports.
[image]

### Project file & Debug log uploads

[Cookie Jar Lid.zip](https://github.com/SoftFever/OrcaSlicer/files/13256707/Cookie.Jar.Lid.zip)
[debug_Sat_Nov_04_08_29_55_49908.log.zip](https://github.com/SoftFever/OrcaSlicer/files/13256710/debug_Sat_Nov_04_08_29_55_49908.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

mydevpeeps, 2023-11-04:
Note: _I just tried this in Bambu Studio v1.7.7.89 and it does not happen there._

igiannakas, 2023-11-04:
Disable the option to make overhang printable.

mydevpeeps, 2023-11-04:
Confirmed. So now the question remains: Is that expected behavior?

kaantuncel, 2024-05-09:
This is still an issue and unchecking the overhang option doesn't solve it

eriko411, 2024-06-11:
This is an issue. I have a white skull with black eyes.  Enabling "make overhang printable" results in the base of the skull printing in black and white which is over riding the colour settings

AndrewLeeVS, 2024-07-01:
Also having this issue

LoghamLogan, 2024-07-15:
This is really annoying behaviour and adds an insane amount of print time for unnessacerry color changes. Only way to stop it is to disable the option and manually add slopes for overhangs onto the model.