OrcaSlicer/OrcaSlicer #2610: add support for sailfish in g-code flavor

myB0B, 2023-11-04:
My old FlashForge Creator Pro still brings me the same satisfaction after years of operation. 
I'm currently using SuperSlicer with gcodeSnippets post-processing script to generate x3g files, but I would like to be able to use Orcaslicer with it but it's missing sailfish (makerbot) g-code flavor in printer settings

Bigbosspt, 2024-01-16:
X2 here, i would like to test it out to !

closed 2024-04-23 by github-actions[bot]

bvaerewyck, 2024-11-12:
Any more interest in this?

myB0B, 2024-11-13:
> Any more interest in this?

Yes ! still using my old FFCP

cordvision, 2024-11-23:
> Any more interest in this?

definitely... also using a Flashforge Creator Pro with Sailfish firmware.

willefg, 2025-01-20:
> Any more interest in this?

Yes, still using my old and modded CTC Bizer with Sailfish.

myB0B, 2025-01-21:
This request has been open for two years, and still nothing. As always, efforts are focused on new products (Bambulab, to name one), but older ones clearly don't interest the developers. Where is the open-source spirit?!

willefg, 2025-01-22:
> This request has been open for two years, and still nothing. As always, efforts are focused on new products (Bambulab, to name one), but older ones clearly don't interest the developers. Where is the open-source spirit?!

To be fair, it's old technology and one would need a capable individual who can create the x3g support to have such an old 3D printer for testing, I assume. Anyone else will weigh their options to spend their time working on something old for a couple of users or something new for the masses. If I had the abilities, I know what I would do.

However, I did find some code designed that, as it seems, is designed to be used to implement in slicer software. Maybe that helps someone.

https://github.com/markwal/GPX

But, since this topic is set to "stale" and "closed as not planned", I won't hold my breath.

myB0B, 2025-01-24:
Not really, I disagree. OrcaSlicer being a fork of PrusaSlicer, and since PrusaSlicer has this feature natively, it was simply "removed."
Maybe it’s as simple as uncommenting a line of code during compilation to make it work, but I’m not a developer.

It’s unfortunate that an open-source tool like this, which brings significant improvements, doesn’t support older hardware. Not everyone has the desire or the budget to buy a BambuLab.
This is yet another strong signal in favor of planned obsolescence and against ecology.

And then there are people, those who just don’t get it, who come and tell you that open source is free, or that you should just do it yourself.

I know GPX, but that's not the point.

phanex, 2025-09-01:
I also have CTC Bizer and would love to use Orca for it.

ingoknito0815, 2025-10-08:
We have a Creator 4 at work and i think we have more or less the same problem. i woul dbe happy if these "older" machines would be supported