OrcaSlicer/OrcaSlicer #2612: Add support for Adventurer 3

ttoille123, 2023-11-04:


RikshaDriver, 2024-02-17:
+1 to this request

JediMasterBob, 2024-02-25:
+1 to this request

RikshaDriver, 2024-03-02:
I've spent a bit of time on this and have created a profile for the Adventurer 3 Series.

See https://github.com/RikshaDriver/adventurer3-orcaslicer-profile

JediMasterBob, 2024-03-04:
thank you @RikshaDriver 

I will give this a try this coming weekend.

RikshaDriver, 2024-03-04:
The Adventurer 3 Series profile has now been merged into the slicer repository and will be available with all Orca Slicer releases moving forward.

@FlashforgeOfficial hopefully you can make necessary adjustments to further enhance this moving forward

JediMasterBob, 2024-03-09:
Hello,

I've added the profile and the printer seems to be working fine. Thanks a lot!

JediMasterBob, 2024-03-09:
I notice that when doing a temp tower, the temp is not actually changing. Is this an issue with the profile setting? Or a limitation of the printer itself?

RikshaDriver, 2024-03-10:
It's likely because the gcode is missing extruder T0 for the command. You will need to manually edit the file. Was looking at scripting this as there are a few commands which require specific T0 extruder identification, but it may be better to apply modifications for identifying dual extruder and fake dual extruder machines. This can then output gcode based on the hardware variance.

JediMasterBob, 2024-03-10:
yes, I see that now. the start G-Code contains the T0, but none of the temp change commands throughout the program have it. 

I will test manually adding the T0, if it works I'll just write a simple script to parse the G-code file to add the T0. Hopefully a future update to Orca will add an option to include the T0 as necessary.

Thanks for your help and for being so responsive!!!

JediMasterBob, 2024-03-10:
Just a couple of other very small items...the thumbnail display isn't working and the time show the run time, not a countdown. I will play with / adjust the thumbnail setting to see if I can get that working. Any idea why the remaining print time countdown is not working?

JediMasterBob, 2024-03-10:
My coding skills kind of suck, but here is what I have so far for adding the T0 on all the M104 lines. I checked the code, and looks good, but not the printer gives a filament error on nearly every layer. In the zip file is also the code that my batch file outputted. Any help would be appreciated!

[Add_T0.zip](https://github.com/SoftFever/OrcaSlicer/files/14551637/Add_T0.zip)

RikshaDriver, 2024-03-12:
> Just a couple of other very small items...the thumbnail display isn't working and the time show the run time, not a countdown. I will play with / adjust the thumbnail setting to see if I can get that working. Any idea why the remaining print time countdown is not working?

The output is gcode and not .gx which is flashforge specific. The .gx file adds a header and uses a bitmap format thumbnail and other parameters which allow the "time remaining" to show on the screen.

I know there are gcode to gx conversion projects out there but I haven't had the time to investigate.

RustyRaindeer, 2024-03-16:
The problem with the temperature tower / M104 is not the T0 missing. The real problem is that there is a comment at the end of the temperature change M104 lines. Just removing thoses comments makes temperature work, as absurd as it sounds!!

closed 2024-06-22 by github-actions[bot]

HellEvro, 2026-09-25:
Late follow-up for the **thumbnail / time-remaining** side of this thread (needs `.gx`, not plain G-code):

Open-source Orca post-processing converter (header + 80×60 BMP, Adventurer profiles included):

https://github.com/HellEvro/FF_Gcode_to_GX/releases/tag/v0.1.0

Drop the binary into **Process → Others → Post-processing scripts**, set output to `*.gx`. Not a substitute for native GX in Orca — just a practical workaround until/if that lands.

RustyRaindeer, 2026-10-09:
> 23\. Late follow-up for the **thumbnail / time-remaining** side of this thread (needs `.gx`, not plain G-code):
>          Open-source Orca post-processing converter (header + 80×60 BMP, Adventurer profiles included):
>          https://github.com/HellEvro/FF_Gcode_to_GX/releases/tag/v0.1.0
>          Drop the binary into **Process → Others → Post-processing scripts**, set output to `*.gx`. Not a substitute for native GX in Orca — just a practical workaround until/if that lands.

Gee, thanks! I just set this up and sent a test job to the printer, the thumbnail now shows, just perfect!