OrcaSlicer/OrcaSlicer #2615: Display time cost (money/h) seperately from filament price (money/kg) in print preview summary

DefenderVex, 2023-11-05:
**Is your feature request related to a problem? Please describe.**
Not exactly a problem but it's a small feature that would make estimating production costs slightly easier. It's definitely a QoL feature.

**Which printers will be beneficial to this feature**
All?

**Describe the solution you'd like**
Simply as described in the title, provide an additional breakdown for the total estimations in the print preview to display the time cost and filament cost separately in addition to the current combined value.

**Describe alternatives you've considered**
I don't think there really are alternatives for this kind of suggestion, I guess you could math it yourself instead of setting the cost per print hour?

**Additional context**
Useful for those who print on a commercial basis.

Ocraftyone, 2023-11-08:
This is already implemented in 1.8 betas

DefenderVex, 2023-11-08:
Where? Because I'm on 1.8.0 Beta 2, it only gives me the combined value.
[image]

Ocraftyone, 2023-11-09:
Whoops! You are right. I will look into adding that metric.

closed 2024-02-21 by github-actions[bot]