OrcaSlicer/OrcaSlicer #2618: Dwell for Bed Heat Soak

CCS86, 2023-11-05:
It is well known that letting your bed heat soak at the desired temperature for some number of minutes leads to more dimensional stability of the bed during probing and printing. This results in much better first layers.

The current behavior seems to launch straight into the print once bed temp is achieved.

If you could add a setting in the filament profile to allow a dwell after bed temp is reached, that would let the user customize this time period.

I use custom g-code in SS and PS to achieve this with my other printer, and use logic to avoid the dwell if the bed is already hot:

```
M140 R[first_layer_bed_temperature]           ; Set bed

{"if heat.heaters[0].current < "}[first_layer_bed_temperature]-1
{"	"}M190 R[first_layer_bed_temperature]                                ; Wait for cold bed
{"	"}G4 S300                                 ; Cold Pause
{"else "}
{"	"}M190 R[first_layer_bed_temperature]                               ; Wait for hot bed
{"	"}G4 S1                                  ; Hot Pause
```

SoftFever, 2023-11-05:
1.8 has added a new feature, you can achieve it by enabling this feature.
You need to define M191 g-code macro by yourself in Klipper though
[image]

CCS86, 2023-11-05:
You could achieve a dwell with that, but it's not the same though.

Plus, I am using this on an X1C, where I cannot define new gcodes.

SoftFever, 2023-11-06:
> You could achieve a dwell with that, but it's not the same though. Plus, I am using this on an X1C, where I cannot define new gcodes.
> […](#)

I'm lost here.
What g-code you want to the slicer to emit if your printer doesn't support the feature?

CCS86, 2023-11-06:
Yeah, I guess this is probably just best implemented in the start g-code. 

Any chance you know to query the current bed temp on an X1C?

closed 2024-02-21 by github-actions[bot]