OrcaSlicer/OrcaSlicer #2620: Default Generic Marlin Machine start g-code

HappyMark93, 2023-11-05:
**Is your feature request related to a problem? Please describe.**
Initial "Default" settings should be safe to run for of shelf and generic printer configurations by setting specific properties of printer.

1. Where is "EXECUTABLE_BLOCK_START" g-code coming from? How to disable?
2. Current default generic marlin machine start g-code overrides M204 from "EXECUTABLE_BLOCK_START'

**Which printers will be beneficial to this feature**
Custom builds, modified configurations, currently not supported of shelf printers.

**Describe the solution you'd like**

1. Line : G1 X205.0 E19 F1000 should be changed  to take in consideration build plate dimensions.
2. option to activate (add to g code output) M201,M203,M204,M205 command should be added to Motion ability tab of printer settings, and default to unchecked box. 
2. Current default generic marlin machine start g-code overrides M204 from "EXECUTABLE_BLOCK_START' it should be fixed by using one or another option.

**Additional context**
My G-code output:
```
; EXECUTABLE_BLOCK_START
M73 P0 R16
M201 X10000 Y10000 Z100 E5000  
M203 X500 Y500 Z10 E60
M204 P10000 R1000 T0 ; sets acceleration (P, T) and retract acceleration (R), mm/sec^2
M205 X8.00 Y8.00 Z0.40 E5.00 ; sets the jerk limits, mm/sec

;TYPE:Custom

G90 ; use absolute coordinates
M83 ; extruder relative mode
M204 S10000 T1000
M104 S220 ; set extruder temp
M140 S60 ; set bed temp
G28 ; home all
```

Current default generic marlin machine start g-code:
```
G90 ; use absolute coordinates
M83 ; extruder relative mode
M204 S[machine_max_acceleration_extruding] T[machine_max_acceleration_retracting]
M104 S[first_layer_temperature] ; set extruder temp
M140 S[first_layer_bed_temperature] ; set bed temp
G28 ; home all
G1 Y1.0 Z0.3 F1000 ; move print head up
M190 S[first_layer_bed_temperature] ; wait for bed temp
M109 S[first_layer_temperature] ; wait for extruder temp
G92 E0.0
; initial load
G1 X205.0 E19 F1000
G1 Y1.6
G1 X5.0 E19 F1000
G92 E0.0
; intro line
G1 Y2.0 Z0.2 F1000
G1 X65.0 E9.0 F1000
G1 X105.0 E12.5 F1000
G92 E0.0
```

mmarcelo-oliveira, 2023-12-11:
Hello, I'm facing the same issue, were you able to find a solution to disabel  EXECUTABLE_BLOCK_START?

HappyMark93, 2024-02-14:
Unfortunately no :(

joe2112, 2024-02-16:
crud what is this thing, I have the same issue

joe2112, 2024-02-16:
ok I stumbled on it, under the printer edit the defaults, on the motion ability tab uncheck emit limits to G-code, if m200 and other speed limit commands are not supported.

PhoenixRion, 2024-04-27:
I unchecked that box but I still get the following.  It is mainly annoying because I want to heat my bed first.
```
; EXECUTABLE_BLOCK_START
M73 P0 R249
M104 S210 ; set nozzle temperature
```

jbjhjm, 2024-06-20:
> I unchecked that box but I still get the following. It is mainly annoying because I want to heat my bed first.
> 
> ```
> ; EXECUTABLE_BLOCK_START
> M73 P0 R249
> M104 S210 ; set nozzle temperature
> ```

It is working for me... M104 is not a part of printer motion configuration.
Try to enter custo gcode on Machine G-code tab -> Machine start G-Code.
If I remember correctly, auto-set-up code will be removed as soon as you enter custom code there.

closed 2024-09-26 by github-actions[bot]