OrcaSlicer/OrcaSlicer #2622: Checking for updates doesn't work on Mac and Windows

szafranski, 2023-11-05:
### OrcaSlicer Version

since circa 1.7

### OS version

Mac Sonoma and Windows 11

### Printer

Prusa i3 and Creality 10S

### How to reproduce

Open an app or check for updates manually

### Actual results

No prompt regarding possible update 

### Expected results

Prompt or info when update is available. It worked few weeks ago but it is not since some time. On both systems

### Project file & Debug log uploads

[log.zip](https://github.com/SoftFever/OrcaSlicer/files/13259836/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

SoftFever, 2023-11-07:
Thank you for reporting this.
Fixed in dev branch

closed 2023-11-07 by SoftFever

seanbrockest, 2024-06-17:
Can we reopen this? Hasn't worked in the last few versions. I just did a clean install on a clean install of windows 10, and it's still broken. Currently using 2.0.0

prsk8er735, 2024-10-31:
> Thank you for reporting this. Fixed in dev branch

Agree with the user above this needs to be reopened. Windows installation v2.1.1 and check for updates does nothing. No prompt no pop update nothing.

FelixBuehler, 2024-11-02:
Can confirm.
I do recall, that i once got a notification for a beta update. Denied that and since then it's broken.

commit 42f8efc0 (2023-11-07): Fixed a regression that checking new version didn't work
Fixed #2622