OrcaSlicer/OrcaSlicer #2623: 1.8.0: Settings Missing from STRENGTH > ADVANCED

CCS86, 2023-11-05:
### OrcaSlicer Version

1.8.0-b2

### OS version

Win 10 x64

### Printer

X1C

### How to reproduce

Look at settings > strength > adanvanced

### Actual results

[image]

These settings are missing and "detect narrow internal solid infill" seems to be forced on.

### Expected results

These settings should be available to change.

### Project file & Debug log uploads

...

### Checklist of files to include

- [ ] Log file
- [ ] Project file

SoftFever, 2023-11-06:
The algorithm has been changed hence these option had been removed.
The ensure vertical shell thickness toggle will be added back in 1.8 official though

Tarakesh83, 2023-11-06:
This forced on option is now actually destroying parts.
Even with Pressure Advance set to 0 in klipper the concentric infill produces gaps where as the old monotonic setting did not.
This behaviour should immediatelly be reverted.

CCS86, 2023-11-06:
> The algorithm has been changed hence these option had been removed. The ensure vertical shell thickness toggle will be added back in 1.8 official though

This is my experience. Loading the same project in 1.8.0-b2, I get a very different and undesirable infill.

[image]

PR #2626 (2023-11-06): Add back the option to disable `ensure vertical shell thickness` and `detect narrow internal solid infill`
Partly reverted commit 075a08bca8a033c896c48f2d3612f74ff7a19716
This should fix #2602 and fix #2623
closed 2023-11-06 by SoftFever