OrcaSlicer/OrcaSlicer #2631: Fuzzy Vertical Seam

will-flaugher, 2023-11-06:
### OrcaSlicer Version

V1.8.0 Beta 2

### OS version

Windows 10 Home

### Additional system information

CPU: Intel Core i5 8600K
Memory: 16GB DDR4
Graphics: NVIDIA GeForce RTX 3070

### Printer

Bambu P1S

### How to reproduce

Set seam position to "random", select your object and begin to paint the seam. Set the brush size as small as possible and then paint the seam. Slice the plate.

### Actual results

[image]

The resulting seam in the preview pane is much fuzzier / less straight than the one drawn onto the model. 

### Expected results

If I'm paint a specific seam into the slicer and using a very fine brush size, I would expect the resulting seam to match it exactly.

### Project file & Debug log uploads

[fuzzy drawn seam.zip](https://github.com/SoftFever/OrcaSlicer/files/13270227/fuzzy.drawn.seam.zip)
[log files.zip](https://github.com/SoftFever/OrcaSlicer/files/13270232/log.files.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

DVSVIDEO, 2023-11-06:
I think that's exactly how it should work. Within the painted line, the seam is randomly placed.
If you choose the "aligned" setting it will be exactly as straight under each other as you painted it, or is that not what you wanted?

will-flaugher, 2023-11-06:
@DVSVIDEO - based on how narrowly I painted the seam line, I was expecting a straight line for this specific section of the print. The scattered output in the preview puts seams outside of the line that I painted.

DVSVIDEO, 2023-11-06:
Yes it is wider than the paint, but I don't think that's really a problem. If you are drawing a very thin line, you can staple it to the straight line and you will get a straight seam.
I don't understand your problem. 
Yes I see, random alignment will also put the starting points slightly outside the painted border, but anyone who chooses random is not bothered by this.
And if you draw so thin then it's hardly random, it's a straight line.

will-flaugher, 2023-11-06:
@DVSVIDEO - regardless of whether I'm choosing nearest/aligned/random for my seam position, when I paint a very straight, very thin line, the slicer should follow this instruction and the resulting seam should also be straight. Deciding on its own to ignore my degree of specificity and put the seam outside of my painted line is a flaw.
If I were to print the object shown in my preview, for example, the seam would come out looking frayed and splintered, rather than nice and aligned.

igiannakas, 2023-11-07:
> @theophile - I need random seams at other parts of the print for functional reasons, but for this smooth surface, I'd like an aligned seam. This is why I'm painting it on and why I'm using such tight tolerances for the seam in my slicer settings.
> 
> It would be nice if I could edit seam positioning with a modifier cube to accomplish this goal, but that's a separate request.
> 
> Regardless of my individual reasons, shouldn't this be my decision rather than the slicer's? If I'm painting a thin, straight line, the slicer should produce a thin, straight seam.

You can change the seam placement on a per object basis.

will-flaugher, 2023-11-07:
@igiannakas am I missing it? I have my modifier cube selected in the list of Objects, but the only place I can adjust the seam position is under the "Global" settings. Below is a screenshot of the settings in the "Object" view.

[image]

DVSVIDEO, 2023-11-07:
> @theophile - I need random seams at other parts of the print for functional reasons, but for this smooth surface, I'd like an aligned seam. This is why I'm painting it on and why I'm using such [...]

I don't think you understand this function of the slicer. This is not a mistake! You cannot use it.
1. If you want to randomly place the seam anywhere on the model, you need the random setting without painting.
2. If you want to randomly place the stitch on the model but scattered within a given band, you need random adjustment with painting
3. If you want it in a defined but continuous line, then you need the adjusted setting
4. If you want it in a defined continuous but completely vertical line, you need to tick the vertical option.

There is no malfunction of any kind.
Feel free to delete this entire post.

DVSVIDEO, 2023-11-07:
It's your fault, not the slicer's.
mine:
[image]

will-flaugher, 2023-11-07:
@DVSVIDEO - and if I want a randomized seam on part of the model, and an aligned seam on another part, I should be able to paint a very fine line for the slicer to follow. It seems like you may have misunderstood my goal.

As far as your screenshot, it looks like you're selecting the entire plate or entire object, rather than a modifier or a part. As soon as you try to add a modifier or split the file into parts, the option to change your seam placement goes away.

DVSVIDEO, 2023-11-07:
Stop!
I quote the title of the post: Fuzzy Vertical Seam...
then you wrote: The resulting seam in the preview pane is much fuzzier / less straight than the one drawn onto the model.

What you are explaining now has nothing to do with the "problem" you raise

Delete this post and stop burdening the programming team with a bug that is not a bug.
Thank you.

will-flaugher, 2023-11-07:
@DVSVIDEO I'm not sure if there's a language barrier or if you're intentionally misunderstanding my concern. If I paint a narrow, straight seam as my input, but the slicer's output is a fuzzy seam that falls outside the borders of my painted line, this is a problem. 

It's no different than choosing a 0.4mm layer height, but the slicer decides to go with 0.2mm instead. Any deviations from the given inputs is an issue.

will-flaugher, 2023-11-08:
@DVSVIDEO - I think we're going to have to agree to disagree and I'll ask you to stop commenting on my bug ticket if you don't have anything constructive to add.

The core of this ticket is, and always has been, that I've painted a very thin, straight line, but the slicer is creating seams outside of my painted line. In cases where software isn't following the explicit instructions of the user, it's a bug.

theophile, 2023-11-08:
I've looked through /src/libslic3r/GCode/SeamPlacer.cpp, and although C is not my strong suit, I think I see the reason for the behavior you're experiencing. It seems to be the combination of two things:

First, the algorithm for selecting candidates for seam placement generally considers a certain number of random points per layer and ranks them according to various criteria. If the point is inside an enforcer it is considered a viable seam point. If it's in a blocker, it is rejected as being a viable candidate. If it's not in either an enforcer or a blocker, it may still be a viable candidate and is ranked as being more or less preferable according to various criteria depending on the selected seam type. When you select "random," the algorithm develops a list of viable seam candidates per layer and then randomly selects one of them. That means that even if the "best" candidate is the one inside the enforcer, it may not be the one chosen because you selected "random." Another way to look at this is that a seam enforcer is not the inverse of a seam blocker. In other words, the algorithm does not automatically reject any candidate that is outside an enforcer.

Second, the model being sliced is processed as a mesh of triangles, and a seam enforcer extends to every part of every triangle that is within the enforcer. If you're using an enforcer that is only 0.05mm wide on a model that has a resolution larger than, say, 0.02mm, there's a good chance that the total horizontal distance covered by all of the mesh polygons intersected by the enforcer will be wider than 0.05mm. 

Your two screenshots aren't the same scale so it's hard to be certain, but by a rough eyeballing, it looks like the width of the seam band is roughly the same as the width of the visible "squares" that make up the curved edge of your model. Assuming those squares are made up of two triangles in the mesh, that would explain why your seam is generated the way it is:

- You paint a 0.05mm wide seam on the model, which is "expanded" to include all parts of every triangle that falls anywhere on that line.
- The seam-generation algorithm flags any seam point within any of those triangles as a viable seam position.
- Because you selected "random," the seam generator selects a random seam point for each layer instead of trying to align them.

If I'm right about this, then what you're seeing is the intended behavior, though I tend to agree with you that there's a potential for user confusion if the "actual" enforcer is materially different in size and shape than the visual representation of it in the GUI. Ideally the painter tool would work by "bucket" filling every polygon it touches, so that the user would be able to see the _true_ bounds of the enforcer. In other words, assume you have a low-poly 10mmx10mmx10mm cube, where each face is a square comprised of two triangles. If you use the seam painter to paint a straight vertical line on a wall, even with a 0.05mm brush, ideally it would "light up" the entire wall because the actual enforcer will include the entirety of the polygons. 

I haven't looked at the painter GUI elements so I'm not sure whether the GUI already does this. If not, I think that would be a good idea to implement to reduce user confusion. If the GUI does already do this, then in your case, that would suggest that your model resolution is already fine enough to achieve the result you want, and I would then check your slicing resolution. The default is 0.01mm, which I think would work well in this scenario, but if yours is set higher than 0.05, that might also explain the behavior you're seeing.

If the GUI does not already "bucket fill" the entire polygons, then your model resolution might be too coarse. You might be able to achieve your desired results by using something like MeshMixer to significantly increase the resolution of your mesh. Or if you have the model in STEP or other CAD form, you could try increasing the resolution of the mesh export.

will-flaugher, 2023-11-09:
@theophile - thanks a lot for the considered reply. Given your comment about scale, I've taken the liberty of adjusting the scale and overlaying the images on top of one another so you can see the difference more clearly. The green line is my painted seam and the white dots are the resulting seams in the "preview" window.

[image]

To your point about resolution, my slicer resolution is currently set to .01mm, but I didn't realize that this could differ from the resolution of the file itself. I'll have to look into that and see if I can increase the resolution of the file as one way to improve the seam's fidelity.

DVSVIDEO, 2023-11-09:
Even after theophile's detailed explanation, if you want to get a thin seam that doesn't spread at all (exactly as you paint), you shouldn't choose the random option.
If you do choose it then (we now know) it is the size of the polygons that causes this phenomenon.
Now you know that this is not a bug, use the "adjusted" option.
I said this at the beginning, but you started to digress and accuse me of not being helpful enough.

theophile, 2023-11-09:
> The green line is my painted seam and the white dots are the resulting seams in the "preview" window.

I took your image and outlined in blue the edges of what appear to me to be the square facets in the model geometry in that area:
[image]
I'm pretty sure this is going to end up being your culprit. If so, increasing the resolution of the mesh itself should help, but you still might not get a perfectly aligned seam while using "random."

will-flaugher, 2023-11-10:
@theophile - yeah, that makes a lot of sense. It would be sweet if it could detect straight, painted lines under a certain width and "understand" that it should be aligned, but at least we have a really solid theory for what's going on. Thanks again for digging in and sharing!

DVSVIDEO, 2023-11-11:
> I took your image and outlined in blue the edges of what appear to me to be the square facets in the model geometry in that area: [image] I'm pretty sure this is going to end up being your culprit. [...]

The "adjusted" point is the opposite of "random."
So why would you want it to be more orderly and less random when you choose to make it "RANDOM".
It makes perfect sense and logic if you choose "ALIGNED" mode and then each starting point will be exactly on top of the other (assuming straight painting, as in the pictures)

theophile, 2023-11-11:
Yes, but he said he needs random seam placement for other parts of the model and it doesn't appear to be possible at the moment to use a modifier to change the seam generator for only a portion of the model. He could make this seam aligned by using the aligned method, but then he would have the random placement he needs in other parts of the model.

DVSVIDEO, 2023-11-11:
Either I'm a wizard or I'm just using the "adjusted" mode:
[image]

DVSVIDEO, 2023-11-11:
This bug is not called a feature error: add a new feature that allows you to set the seam type while using the modifier.

discip, 2023-11-18:
@will-flaugher
If your issue is solved, please don't forget to close it. 😊

thanks

will-flaugher, 2023-11-18:
@discip - no worries, the issue isn't solved. DVSVIDEO doesn't seem to understand it, but theophile was very helpful in pointing out the likely reason that it's not working as it should.

DVSVIDEO, 2023-11-19:
Why do you say I don't understand?

The error message is called Fuzzy Vertical Seam ( this means that the vertical seam is scattered...)

Then you wrote this:
How to reproduce
Set seam position to "random", select your object and begin to paint the seam. Set the brush size as small as possible and then paint the seam. Slice the plate.

So the mistake you think is that if you set it to random, then setting the brush to the smallest size will not make the seam straight, it will spread slightly horizontally...

So far I have understood correctly????

In my answer I explained and illustrated with a picture that it is enough to set it to "aligned" mode and then even with a thicker brush it will not be fuzzy...

If this is not the real problem, why is this the title, why did you insert such a picture and why did you write the order of the error as you did?

If the problem is different, then close this and open a new one where you name the real problem in the title and paste a picture that fits it...

To address the problem in a meaningful way, the developers need to be able to explain it well.

closed 2024-02-25 by github-actions[bot]