OrcaSlicer/OrcaSlicer #2641: Internal Bridge Missing if Sparse Infill is One Layer Thick

ra1nb0wtrout, 2023-11-07:
### OrcaSlicer Version

1.8beta2

### OS version

Windows

### Printer

Rat Rig V-Core 3.1

### How to reproduce

Slice a thin part, which would result with a single layer of sparse infill between the bottom and top internal solid fill

### Actual results

Solid internal infill is generated directly on top of the sparse infill, without any internal bridge.

This means the solid fill doesn't have anything to anchor to at the ends.

### Expected results

Sparse infill is replaced with the proper internal bridge, or converted to internal solid fill.

As a result, the top internal solid fill has something more substantial to be anchored on.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Noisyfox, 2023-11-07:
Could you upload a test project file that can reproduce this issue?

ra1nb0wtrout, 2023-11-07:
[bridge_example.zip](https://github.com/SoftFever/OrcaSlicer/files/13281571/bridge_example.zip)

Here's an example, see layers 4 and 5, right in the middle of the part.

Noisyfox, 2023-11-12:
This issue is caused by the fact that the bridge is too thick to fit into the sparse infill layer, which means it doesn't need to be a bridge layer at all, since this layer will stick to the layer below regardless. In your case, layer 4 is thinner than the bridge height, so that layer 5 will stick on to layer 3 so no bridge is really needed.

closed 2024-02-21 by github-actions[bot]