OrcaSlicer/OrcaSlicer #2643: FR: Add ability to select sidebar location (left or right)

mlewis-everley, 2023-11-07:
**Is your feature request related to a problem? Please describe.**
For some reason Bambu Labs decided to switch the default Slic3R sidebar location from the right hand side to the left hand side. But, at least, in the left to right hand reading  parts of the world, this UI change can be very obstructive.

In a left to right reading workflow, the left side of the UI should be focused on the tasks you will be doing most in the software, which (at least for myself) involves interacting with the buildplate(3) and objects located on it. Additional settings contained in the sidebar are very much a secondary activity (which should be on the right).

It is not the end of the world, but I do find it very distracting having the sidebar to the left. I find my eye is constantly drawn to it when I am trying to work on the buildplate.

**Which printers will be beneficial to this feature**
All

**Describe the solution you'd like**
I would love to have the ability to select if the sidebar is rendered on the left or the right. This seems like the best option as it allows users to choose which layout works best for them.

I guess it would be best if this was rendered as a toggle button (or dropdown) in the preferences window?

**Describe alternatives you've considered**
I guess, moving the bar back to the right hand side :-)

**Additional context**
Add any other context or screenshots about the feature request here.

discip, 2023-12-08:
@mlewis-everley
This is addressed by #2972?
Please consider closing this issue. 😊

mlewis-everley, 2023-12-09:
Looks to be, thanks for letting me know 🙂