OrcaSlicer/OrcaSlicer #2644: Filament change after purging

stdakov, 2023-11-07:
### OrcaSlicer Version

1.7.0

### OS version

macOS

### Additional system information

M2

### Printer

Kingroon KP3S S1

### How to reproduce

Add change filament

### Actual results

It adds M600 command right after purging before starting printing

### Expected results

To not add M600 before starting printing.

### Project file & Debug log uploads

[ladybug_PLA_26687994d0h0m.gcode.zip](https://github.com/SoftFever/OrcaSlicer/files/13279962/ladybug_PLA_26687994d0h0m.gcode.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

olcayoren, 2023-11-07:
Hello, I also reported a similar problem. When you change the filament, the slicer completion time increases dramatically and thus printing problems occur, Gcode is interrupted. Do you also experience a deterioration in the completion time?

[https://github.com/SoftFever/OrcaSlicer/issues/2635](url)

stdakov, 2023-11-07:
No but I am facing another issue, first two layer after changing the filament, the extruder does not work

olcayoren, 2023-11-07:
> No but I am facing another issue, first two layer after changing the filament, the extruder does not work

We are experiencing the same problem. For example, if the filename is 26687994d0h0m.gcode instead of 8m50s.gcode, the printer does not continue after the filament change and gives an error. In fact, the fact that the file name is long indicates that some parts of the gcode are incorrect.

SoftFever, 2023-11-09:
I've pushed a fix for the issue.

Could you help verify if the fix works by trying the latest dev build?

closed 2023-11-12 by SoftFever