OrcaSlicer/OrcaSlicer #2645: Slicer creating geometry inside infill underneath the actual geometry

msteele999, 2023-11-07:
### OrcaSlicer Version

1.8.0 beta 2

### OS version

Windows 11 Home

### Printer

Bambu Lab X1cc

### How to reproduce

- import model
- change filament to something easy to view (yellow)
- slice
- view layers and notice that they show infill, then some geometry, then infill (see screen shots)
- this was not happening in 1.7.0
- turning 'Only one wall on top surfaces' to off changes the behavior
- These are sliced with 100% infill to highlight what is happening

First two images are v1.7.0 - remaining images are 1.8.0 beta 2 (also happened in beta 2)

[image]

[image]

1.8.0 beta 2

[image]
[image]
[image]
[image]
[image]

### Actual results

this creates issues when printing - lots of retractions and slower printing due to printing geometry that will never be seen and is not needed for support.

### Expected results

just print smooth 100% infill layers until the actual geometry is reached.

I have inspected the model in Fusion using sectional analysis and the geometry is not overlapping - they sketches are created on top of the main surface and extruded upwards.

### Project file & Debug log uploads

[log.zip](https://github.com/SoftFever/OrcaSlicer/files/13281126/log.zip)
[geometry_inside_infill.zip](https://github.com/SoftFever/OrcaSlicer/files/13281127/geometry_inside_infill.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

msteele999, 2023-11-08:
Update - if I increase the number of bottom layers to be equal to the layer immediately before the color change / where the geometry SHOULD start, I can get the slice to only have ONE layer of geometry in the infill - which is workable.

[image]

Noisyfox, 2023-11-12:
When I open the 3mf it says the file does not contain any geometry data. Could you reload the project file?

msteele999, 2023-11-12:
I saved the project again - please let me know if this one works.

[RF Radiation - New v4.zip](https://github.com/SoftFever/OrcaSlicer/files/13328032/RF.Radiation.-.New.v4.zip)

Noisyfox, 2023-11-12:
Yep that works.

For now you can disable the "Detect narrow internal solid infill" and "Ensure vertical shell thickness" which should greatly improve the quality (but the "Only one wall on top surfaces" still won't work well).

msteele999, 2023-11-12:
Will do - standby

msteele999, 2023-11-12:
Ouch - that actually made it worse.

Making the bottom layers >= the start of the new geometry works the best.

Noisyfox, 2023-11-12:
Yeah I noticed. Your model is actually a great benchmark for testing slicing thing but complicated objects, and I've so far noticed a few more other types of defects, comparing to BBS and SS. Will try fixing each of them.

msteele999, 2023-11-12:
Nice!

Glad to assist - let me know how I can contribute

Noisyfox, 2023-11-13:
Are you able to give this build a try and see if that improves the quality? https://github.com/SoftFever/OrcaSlicer/actions/runs/6846552190

msteele999, 2023-11-13:
This works properly!

msteele999, 2023-11-13:
Note - I still have to set the infill to 100% - if I set it lower - I see
the geometry inside the lower layers still - but I can work around that

msteele999, 2023-11-13:
> Yeah I'm aware of this. Those are different issues though and I'm still
> investigating.

Very good!

Please let me know how I can assist.

Noisyfox, 2023-11-13:
Another tip:
Set "One wall threshold" to 0 while "Only one wall on top surfaces" turned on will remove the scars on the surface.

msteele999, 2023-11-13:
testing!

msteele999, 2023-11-13:
Which of these is one wall threshold? I assume by '0' you mean unchecked?

msteele999, 2023-11-13:
[image: image.png]

msteele999, 2023-11-13:
Nevermind - I see it - have to select it first to get the threshold

msteele999, 2023-11-13:
That still leaves one layer with the next geometry being printed even
though it should be a solid top layer :-(

[image: image.png]

Noisyfox, 2023-11-14:
Hmmmm unfortunately your images are all broken.

closed 2023-11-29 by SoftFever

scruplelesswizard, 2024-07-10:
> Another tip: Set "One wall threshold" to 0 while "Only one wall on top surfaces" turned on will remove the scars on the surface.

Setting _One wall threshold_ to `0` solved issues I was having in 2.1.1 with unexpected infill patterns under text. Does this behavior indicate a regression for #2716?

### `One wall threshold=300%`
[image]

### `One wall threshold=0`
[image]

PR #2716 (2023-11-29): Various infill improvements
1. Fix issue that sparse infill threshold no longer working
2. Turn all internal sparse infill into solid infill if infill density is 100%
3. Allow combining solid infill when sparse infill density is 100%
4. Move internal solid infill pattern settings into the Infill section

This should fix #2645