OrcaSlicer/OrcaSlicer #2646: When new curled overhang feature is enabled, overhang perimeter speeds are not respected

stew675, 2023-11-07:
### OrcaSlicer Version

1.8.0 Beta 2

### OS version

Windows 11

### Additional system information

Not relevant

### Printer

Generic

### How to reproduce

Download test model from here: https://www.thingiverse.com/thing:6302570

Enable curled overhang option Orca Slicer, which also involves disabling "Classic Mode" for Overhang Speed.

In the filament cooling settings, set max fan speed threshold layer time to 2s. Also ensure that "Slow printing down for better layer cooling" is enabled.

In speed settings, set perimeter speed to 200mm/s.  Keep the overhang speed settings at their defaults (0, 50, 30, 10)

Now slice the part and look at the perimeter speeds.

### Actual results

It can be observed that the perimeter speeds on the overhangs are significantly higher than the overhang speed settings.

If the filament cooling max fan speed threshold layer time is increase and the model is resliced, we can observe that the overhang perimeter speeds are tracking the speeds as determined by the minimum layer time only.

Additionally, if the  "Slow printing down for better layer cooling"  option is disabled, and the model is resliced, we can observe that the overhang perimeter speeds are generated at full speed.

### Expected results

We should be expecting to see overhang perimeter speeds respect the speeds set for the respective overhang percentages, but we do not.

If we re-enable Classic Mode for over-hang speed, we can see that the overhang perimeter speeds are respected again.

### Project file & Debug log uploads

None

### Checklist of files to include

- [ ] Log file
- [ ] Project file

igiannakas, 2023-11-11:
Can you attach your 3mf project file please?

sleewok, 2023-12-28:
I can't get slowdown for curling to have an impact. I tried on 1.8 and 1.9alpha

igiannakas, 2023-12-28:
Submit your 3mf project file please.

closed 2024-04-05 by github-actions[bot]