OrcaSlicer/OrcaSlicer #2654: Option to wipe nozzle with # of wipes at set # of layers

skl111, 2023-11-08:
**Is your feature request related to a problem? Please describe.**
When printing PETG depending on settings and ooze, a nozzle wipe maybe needed at every layer or every few layers.

**Which printers will be beneficial to this feature**
X1 Series, P1 Series, A1 Series

**Describe the solution you'd like**
Option to enable Nozzle wiping with # of wipes at each layer or each X number of layers

**Describe alternatives you've considered**
I have found g-code that may work, but I have not tested it yet.

**Additional context**
Add any other context or screenshots about the feature request here.

skl111, 2023-12-11:
Adding to this, have this setting available per object as well.

VirtualEnder, 2024-01-23:
I came here to suggest this, but you already did. Definitely would make printing PETG easier. Even with completely dry filament I get buildup on the nozzle that sometimes drops off into the print. Sometimes it comes off in the infill, which is fine, but sometimes it catches on an outer edge and causes cosmetic issues with the print. Also sometimes it drags and causes visible issues where it prevents the deposited filament from going down smoothly. Adding an option to extrude a small bit of filament and do an nozzle wipe on the cleaner every X number of layers would be awesome. Seems like not a hard thing to add either. @skl111 Can you share the GCode you found?  I may have a whack at the implementation. If not I can try to see if I can find the generated gcode for the nozzle wipe during the initial sequence.

VirtualEnder, 2024-01-23:
So I dug through the machine GCode, as well as the gui code, and I think I can take a whack at implementing this.  I will fork Orca and see if I can get it working in the gui. If I can get it working reliably I'll submit a pull request.

skl111, 2024-01-24:
Hey, So here is the code I run manually at the 2nd layer and then depending on when needed I re-insert it.

;move up .4 to clear prints
G1 Z{layer_z + 0.4}
;move to wipe area
G1 X60 Y265 F10000
G1 X100 F5000
G1 X60 F10000
G1 Y264 F10000
G1 X100 F5000
G1 X60 F10000
G1 Y263 F10000
G1 X100 F5000
G1 X60 F10000
G1 Y264 F10000
G1 X100 F5000
G1 X60 F10000
G1 Y265 F10000
G1 X100 F5000
;G1 X60 F10000
G1 Y260 F10000
G1 X127 F5000

And thats because I use the original Wiper and and A1 and A1 mini wiper together and the Y changes to hit the different A1 brush nodules.

https://makerworld.com/en/models/94607#profileId-101194

I would suggest to make sure this is in the advanced options and have and option to use the default nozzle wipe code, or a modified nozzle wipe code such as the one I use.

closed 2024-05-02 by github-actions[bot]

erwann17, 2024-10-27:
hello,
 I'm trying to do the same thing, that is to say introduce a gcode every 5 layers.
 have you found a solution