OrcaSlicer/OrcaSlicer #2656: Not notifying anymore about new releases

martinrosolino, 2023-11-08:
### OrcaSlicer Version

1.8.0 beta 1

### OS version

W11

### Printer

Hellbot Magna 2 300

### How to reproduce

Some versions before, i'll pop a message saying that a new version was abailable, what happened?

### Actual results

No notification about nwe releases

### Expected results

Notification about nwe releases

### Project file & Debug log uploads

idk

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Ocraftyone, 2023-11-08:
It is possible that it is because the new releases are betas.

martinrosolino, 2023-11-08:
ok, I don't remember since when I've stopped receiving notifications, but it may be the cause, thanks man

discip, 2023-11-18:
@martinrosolino
As this seems to be solved, please don't forget to close. 😊

thanks

closed 2023-11-18 by martinrosolino