OrcaSlicer/OrcaSlicer #2660: MK3S preset can not be loaded

Charger93, 2023-11-08:
### OrcaSlicer Version

1.8.0 Beta 1 and 2

### OS version

Windows 10 and 11

### Additional system information

Not important

### Printer

Prusa MK3S+

### How to reproduce

1. Start the OrcaSlicer without existing configuration
2. Select the MK3S as the used printer in the wizard

Now, the Slicer only shows the default printer.

### Actual results

OrcaSlicer don't use the profile

### Expected results

The Slicer should use the profile.

How to fix: remove the "draft_shield = 0" from the "process_common_mk3.json" and it works!
Maybe the variabe must be renamed, I don't know. For me, the profile now works.

### Project file & Debug log uploads

2023-11-08 08:27:51.449158[Thread 0x000008e8]:Installing 1 bundles from resources ...
2023-11-08 08:27:51.451158[Thread 0x000008e8]:[BBL Updater]:Performing 4 updates
2023-11-08 08:27:51.451158[Thread 0x000008e8]:	Update(D:\EigeneDateien\Downloads\OrcaSlicer_Windows_V1.8.0-beta2_portable\resources\profiles\Prusa.json -> C:\Users\xyz\AppData\Roaming\OrcaSlicer\system\Prusa.json)
2023-11-08 08:27:51.460259[Thread 0x000008e8]:	Update(D:\EigeneDateien\Downloads\OrcaSlicer_Windows_V1.8.0-beta2_portable\resources\profiles\Prusa\process -> C:\Users\xyz\AppData\Roaming\OrcaSlicer\system\Prusa\process)
2023-11-08 08:27:51.460259[Thread 0x000008e8]:copy_directory_recursively "D:\EigeneDateien\Downloads\OrcaSlicer_Windows_V1.8.0-beta2_portable\resources\profiles\Prusa\process" -> "C:\Users\xyz\AppData\Roaming\OrcaSlicer\system\Prusa\process"
2023-11-08 08:27:51.571630[Thread 0x000008e8]:	Update(D:\EigeneDateien\Downloads\OrcaSlicer_Windows_V1.8.0-beta2_portable\resources\profiles\Prusa\filament -> C:\Users\xyz\AppData\Roaming\OrcaSlicer\system\Prusa\filament)
2023-11-08 08:27:51.571630[Thread 0x000008e8]:copy_directory_recursively "D:\EigeneDateien\Downloads\OrcaSlicer_Windows_V1.8.0-beta2_portable\resources\profiles\Prusa\filament" -> "C:\Users\xyz\AppData\Roaming\OrcaSlicer\system\Prusa\filament"
2023-11-08 08:27:51.709693[Thread 0x000008e8]:	Update(D:\EigeneDateien\Downloads\OrcaSlicer_Windows_V1.8.0-beta2_portable\resources\profiles\Prusa\machine -> C:\Users\xyz\AppData\Roaming\OrcaSlicer\system\Prusa\machine)
2023-11-08 08:27:51.709693[Thread 0x000008e8]:copy_directory_recursively "D:\EigeneDateien\Downloads\OrcaSlicer_Windows_V1.8.0-beta2_portable\resources\profiles\Prusa\machine" -> "C:\Users\xyz\AppData\Roaming\OrcaSlicer\system\Prusa\machine"
2023-11-08 08:27:51.772190[Thread 0x000008e8]:No bundles need to be removed
2023-11-08 08:27:51.772190[Thread 0x000008e8]:Slic3r::PresetBundle::load_presets enter, substitution_rule 4, preferred printer_model_id MK3S
2023-11-08 08:27:51.772190[Thread 0x000008e8]:Slic3r::PresetBundle::load_vendor_configs_from_json: parse C:\Users\xyz\AppData\Roaming\OrcaSlicer\system/Custom.json, got description:  "My configurations"
2023-11-08 08:27:51.809971[Thread 0x000008e8]:Slic3r::PresetBundle::load_vendor_configs_from_json: parse C:\Users\xyz\AppData\Roaming\OrcaSlicer\system/Prusa.json, got description:  "Prusa configurations"
2023-11-08 08:27:51.825602[Thread 0x000008e8]:Slic3r::ConfigBase::load_from_json: parse C:\Users\xyz\AppData\Roaming\OrcaSlicer\system/Prusa/process/process_common_mk3.json got a generic exception, reason = Invalid value provided for parameter draft_shield: 0
2023-11-08 08:27:51.825602[Thread 0x000008e8]:Slic3r::PresetBundle::load_vendor_configs_from_json::<lambda_4760ed3e8311b28387eaf1ce8ddb8f34>::operator (): load config file C:\Users\xyz\AppData\Roaming\OrcaSlicer\system/Prusa/process/process_common_mk3.json Failed!
2023-11-08 08:27:51.825602[Thread 0x000008e8]:Slic3r::PresetBundle::load_vendor_configs_from_json, got error when parse process setting from C:\Users\xyz\AppData\Roaming\OrcaSlicer\system/Prusa/process/process_common_mk3.json

### Checklist of files to include

- [ ] Log file
- [ ] Project file

blooest, 2023-11-11:
Every other instance of that key I see across the profiles suite shows "disabled" rather than "0". Probably just needs to be changed to match.

SoftFever, 2023-11-12:
Thank you very much.
Fixed

commit 60424baa (2023-11-12): Fix draft_shield error for MK3S profile
Fixed #2660
closed 2023-11-12 by SoftFever