OrcaSlicer/OrcaSlicer #2661: CLI slice - Segmentation fault

vtl-btz, 2023-11-08:
### OrcaSlicer Version

1.6.6, 1.7.0, 1.8.0 Beta2

### OS version

Ubuntu 20.04, ArchLinux, Windows 10

### Printer

Bambulab X1 Carbon 

### How to reproduce

```
orca-slicer --debug 6 --slice 0  --export-3mf ./ouput.3mf ~/3dprint_files/micro_sim_adapter.3mf 
```

### Actual results

Segmentation fault (core dumped)
RETURN 139

### Expected results

Exit without errors + sliced ouput.3mf

### Project file & Debug log uploads

```
[2023-11-08 11:08:47.582104] [0x00007f2c778d1600] [trace]   Initializing StaticPrintConfigs
[2023-11-08 11:08:47.590777] [0x00007f2c778d1600] [info]    start_gui=false

[2023-11-08 11:08:47.590801] [0x00007f2c778d1600] [info]    Will start to read model file now, file count :1

[2023-11-08 11:08:47.590833] [0x00007f2c778d1600] [info]    plate_to_slice=0, normative_check=1
[2023-11-08 11:08:47.590847] [0x00007f2c778d1600] [info]    read model file:/home/some_user/3dprint_files/micro_sim_adapter.3mf

[2023-11-08 11:08:47.590904] [0x00007f2c778d1600] [info]    model 0x7ffd38a1c0e0, id 3, backup_path empty, set to /tmp/orcaslicer_model/Wed_Nov_08/11_08_47#12201#3
[2023-11-08 11:08:47.590921] [0x00007f2c778d1600] [info]    create /3D/Objects in "/tmp/orcaslicer_model/Wed_Nov_08/11_08_47#12201#3"
[2023-11-08 11:08:47.590963] [0x00007f2c778d1600] [info]    create /Metadata in "/tmp/orcaslicer_model/Wed_Nov_08/11_08_47#12201#3"
[2023-11-08 11:08:47.590994] [0x00007f2c778d1600] [info]    create /lock.txt in "/tmp/orcaslicer_model/Wed_Nov_08/11_08_47#12201#3"
[2023-11-08 11:08:47.591046] [0x00007f2c778d1600] [info]    _load_model_from_file:1495import 3mf IMPORT_STAGE_OPEN, m_load_restore=0

[2023-11-08 11:08:47.591105] [0x00007f2c778d1600] [info]    _load_model_from_file:1531import 3mf IMPORT_STAGE_READ_FILES

[2023-11-08 11:08:47.593571] [0x00007f2c778d1600] [info]    load_3mf found metadata key = Application, value = BambuStudio-01.07.04.52
[2023-11-08 11:08:47.593585] [0x00007f2c778d1600] [info]    load_3mf found metadata key = BambuStudio:3mfVersion, value = 1
[2023-11-08 11:08:47.593593] [0x00007f2c778d1600] [trace]   design_info, load_3mf found copyright = 
[2023-11-08 11:08:47.593598] [0x00007f2c778d1600] [info]    load_3mf found metadata key = CopyRight, value = 
[2023-11-08 11:08:47.593604] [0x00007f2c778d1600] [info]    load_3mf found metadata key = CreationDate, value = 2023-11-08
[2023-11-08 11:08:47.593613] [0x00007f2c778d1600] [trace]   design_info, load_3mf found description = 
[2023-11-08 11:08:47.593623] [0x00007f2c778d1600] [info]    load_3mf found metadata key = Description, value = 
[2023-11-08 11:08:47.593640] [0x00007f2c778d1600] [trace]   design_info, load_3mf found designer = 
[2023-11-08 11:08:47.593652] [0x00007f2c778d1600] [info]    load_3mf found metadata key = Designer, value = 
[2023-11-08 11:08:47.593664] [0x00007f2c778d1600] [trace]   design_info, load_3mf found designer_cover = 
[2023-11-08 11:08:47.593670] [0x00007f2c778d1600] [info]    load_3mf found metadata key = DesignerCover, value = 
[2023-11-08 11:08:47.593677] [0x00007f2c778d1600] [trace]   design_info, load_3mf found designer_user_id = 
[2023-11-08 11:08:47.593681] [0x00007f2c778d1600] [info]    load_3mf found metadata key = DesignerUserId, value = 
[2023-11-08 11:08:47.593688] [0x00007f2c778d1600] [trace]   design_info, load_3mf found license = 
[2023-11-08 11:08:47.593692] [0x00007f2c778d1600] [info]    load_3mf found metadata key = License, value = 
[2023-11-08 11:08:47.593699] [0x00007f2c778d1600] [info]    load_3mf found metadata key = ModificationDate, value = 2023-11-08
[2023-11-08 11:08:47.593705] [0x00007f2c778d1600] [trace]   design_info, load_3mf found model name = 
[2023-11-08 11:08:47.593710] [0x00007f2c778d1600] [info]    load_3mf found metadata key = Origin, value = 
[2023-11-08 11:08:47.593715] [0x00007f2c778d1600] [trace]   design_info, load_3mf found model name = 
[2023-11-08 11:08:47.593719] [0x00007f2c778d1600] [info]    load_3mf found metadata key = Title, value = 
[2023-11-08 11:08:47.593789] [0x00007f2c778d1600] [info]    _load_model_from_file:1682extract 1th file [Content_Types].xml, total=12

[2023-11-08 11:08:47.593805] [0x00007f2c778d1600] [info]    _load_model_from_file:1682extract 2th file Metadata/plate_1.png, total=12

[... 49 lines cut ...]
[2023-11-08 11:08:47.598895] [0x00007f2c778d1600] [info]    total 1 models, 1 objects

[2023-11-08 11:08:47.600041] [0x00007f2c778d1600] [debug]   set_index: plate_id update from -1052334412 to 0
[2023-11-08 11:08:47.600056] [0x00007f2c778d1600] [debug]   set_index: plate_id update from 0 to 1
[2023-11-08 11:08:47.600093] [0x00007f2c778d1600] [info]    reset_size:before size: plate_width 0, plate_depth 0, plate_height 0
[2023-11-08 11:08:47.600107] [0x00007f2c778d1600] [info]    reset_size:after size: plate_width 256, plate_depth 256, plate_height 250
[2023-11-08 11:08:47.600151] [0x00007f2c778d1600] [info]    set_pos_and_size: plate_id 0, before, origin {0,0,0}, plate_width 0, plate_depth 0, plate_height 0
[2023-11-08 11:08:47.600172] [0x00007f2c778d1600] [info]    set_pos_and_size: with_instance_move 0, after, origin {0,0,0}, plate_width 256, plate_depth 256, plate_height 250
[2023-11-08 11:08:47.600187] [0x00007f2c778d1600] [info]    update_slice_result_valid_state: plate 0 , update slice result from 0 to 0
[2023-11-08 11:08:47.600207] [0x00007f2c778d1600] [info]    set_pos_and_size: plate_id 1, before, origin {0,0,0}, plate_width 0, plate_depth 0, plate_height 0
[2023-11-08 11:08:47.600229] [0x00007f2c778d1600] [info]    set_pos_and_size: with_instance_move 0, after, origin {307.2,0,0}, plate_width 256, plate_depth 256, plate_height 250
[2023-11-08 11:08:47.600243] [0x00007f2c778d1600] [info]    update_slice_result_valid_state: plate 1 , update slice result from 0 to 0
[2023-11-08 11:08:47.600258] [0x00007f2c778d1600] [info]    update_slice_result_valid_state: plate 0 , update slice result from 0 to 0
[2023-11-08 11:08:47.600271] [0x00007f2c778d1600] [info]    update_slice_result_valid_state: plate 1 , update slice result from 0 to 0
Segmentation fault (core dumped)
RETURN 139

```

[micro_sim_adapter.3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/13294566/micro_sim_adapter.3mf.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

closed 2023-11-09 by SoftFever

vtl-btz, 2023-11-16:
Still catch segfault, tested on OrcaSlicer_Linux_V1.8.0-rc2 even after resaving project file.
```
$ ./AppRun --debug 6 --slice 0 --no-check  --export-3mf ~/ouput_new.3mf ~/3dprint_files/micro_sim_adapter_new.3mf 
[2023-11-16 11:01:52.087371] [0x00007ff4db31a600] [trace]   Initializing StaticPrintConfigs
[2023-11-16 11:01:52.097356] [0x00007ff4db31a600] [info]    start_gui=false

[2023-11-16 11:01:52.097381] [0x00007ff4db31a600] [info]    Will start to read model file now, file count :1

[2023-11-16 11:01:52.097413] [0x00007ff4db31a600] [info]    plate_to_slice=0, normative_check=1
[2023-11-16 11:01:52.097423] [0x00007ff4db31a600] [info]    read model file:/home/some_user/3dprint_files/micro_sim_adapter_new.3mf

[2023-11-16 11:01:52.097491] [0x00007ff4db31a600] [info]    model 0x7fffc69682a0, id 3, backup_path empty, set to /tmp/orcaslicer_model/Thu_Nov_16/11_01_52#30565#3
[2023-11-16 11:01:52.097509] [0x00007ff4db31a600] [info]    create /3D/Objects in "/tmp/orcaslicer_model/Thu_Nov_16/11_01_52#30565#3"
[2023-11-16 11:01:52.097555] [0x00007ff4db31a600] [info]    create /Metadata in "/tmp/orcaslicer_model/Thu_Nov_16/11_01_52#30565#3"
[2023-11-16 11:01:52.097571] [0x00007ff4db31a600] [info]    create /lock.txt in "/tmp/orcaslicer_model/Thu_Nov_16/11_01_52#30565#3"
[2023-11-16 11:01:52.097616] [0x00007ff4db31a600] [info]    _load_model_from_file:1501import 3mf IMPORT_STAGE_OPEN, m_load_restore=0

[2023-11-16 11:01:52.097669] [0x00007ff4db31a600] [info]    _load_model_from_file:1537import 3mf IMPORT_STAGE_READ_FILES

[2023-11-16 11:01:52.100856] [0x00007ff4db31a600] [info]    load_3mf found metadata key = Application, value = BambuStudio-01.07.07.89
[2023-11-16 11:01:52.100875] [0x00007ff4db31a600] [info]    load_3mf found metadata key = BambuStudio:3mfVersion, value = 1
[2023-11-16 11:01:52.100884] [0x00007ff4db31a600] [trace]   design_info, load_3mf found Copyright = 
[2023-11-16 11:01:52.100889] [0x00007ff4db31a600] [info]    load_3mf found metadata key = Copyright, value = 
[2023-11-16 11:01:52.100910] [0x00007ff4db31a600] [info]    load_3mf found metadata key = CreationDate, value = 2023-11-16
[2023-11-16 11:01:52.100917] [0x00007ff4db31a600] [trace]   design_info, load_3mf found description = 
[2023-11-16 11:01:52.100921] [0x00007ff4db31a600] [info]    load_3mf found metadata key = Description, value = 
[2023-11-16 11:01:52.100928] [0x00007ff4db31a600] [trace]   design_info, load_3mf found designer = 
[2023-11-16 11:01:52.100932] [0x00007ff4db31a600] [info]    load_3mf found metadata key = Designer, value = 
[2023-11-16 11:01:52.100938] [0x00007ff4db31a600] [trace]   design_info, load_3mf found designer_cover = 
[2023-11-16 11:01:52.100942] [0x00007ff4db31a600] [info]    load_3mf found metadata key = DesignerCover, value = 
[2023-11-16 11:01:52.100949] [0x00007ff4db31a600] [trace]   design_info, load_3mf found designer_user_id = 
[2023-11-16 11:01:52.100953] [0x00007ff4db31a600] [info]    load_3mf found metadata key = DesignerUserId, value = 
[2023-11-16 11:01:52.100959] [0x00007ff4db31a600] [trace]   design_info, load_3mf found license = 
[2023-11-16 11:01:52.100963] [0x00007ff4db31a600] [info]    load_3mf found metadata key = License, value = 
[2023-11-16 11:01:52.100970] [0x00007ff4db31a600] [info]    load_3mf found metadata key = ModificationDate, value = 2023-11-16
[2023-11-16 11:01:52.100976] [0x00007ff4db31a600] [trace]   design_info, load_3mf found model name = 
[2023-11-16 11:01:52.100980] [0x00007ff4db31a600] [info]    load_3mf found metadata key = Origin, value = 
[2023-11-16 11:01:52.100986] [0x00007ff4db31a600] [trace]   design_info, load_3mf found model name = 
[2023-11-16 11:01:52.100990] [0x00007ff4db31a600] [info]    load_3mf found metadata key = Title, value = 
[2023-11-16 11:01:52.101073] [0x00007ff4db31a600] [info]    _load_model_from_file:1688extract 1th file [Content_Types].xml, total=14
[2023-11-16 11:01:52.101085] [0x00007ff4db31a600] [info]    _load_model_from_file:1778, [Content_Types].xml skipped, already parsed or a directory or not supported

[2023-11-16 11:01:52.101096] [0x00007ff4db31a600] [info]    _load_model_from_file:1688extract 2th file Metadata/plate_1.png, total=14
[2023-11-16 11:01:52.101183] [0x00007ff4db31a600] [info]    _extract_file_from_archive, extract  "/tmp/orcaslicer_model/Thu_Nov_16/11_01_52#30565#3/Metadata/plate_1.png" from 3mf Metadata/plate_1.png, ret 1

[2023-11-16 11:01:52.101197] [0x00007ff4db31a600] [info]    _load_model_from_file:1688extract 3th file Metadata/plate_1_small.png, total=14
[2023-11-16 11:01:52.101247] [0x00007ff4db31a600] [info]    _extract_file_from_archive, extract  "/tmp/orcaslicer_model/Thu_Nov_16/11_01_52#30565#3/Metadata/plate_1_small.png" from 3mf Metadata/plate_1_small.png, ret 1

[2023-11-16 11:01:52.101259] [0x00007ff4db31a600] [info]    _load_model_from_file:1688extract 4th file Metadata/top_1.png, total=14
[2023-11-16 11:01:52.101304] [0x00007ff4db31a600] [info]    _extract_file_from_archive, extract  "/tmp/orcaslicer_model/Thu_Nov_16/11_01_52#30565#3/Metadata/top_1.png" from 3mf Metadata/top_1.png, ret 1

[2023-11-16 11:01:52.101316] [0x00007ff4db31a600] [info]    _load_model_from_file:1688extract 5th file Metadata/pick_1.png, total=14
[2023-11-16 11:01:52.101358] [0x00007ff4db31a600] [info]    _extract_file_from_archive, extract  "/tmp/orcaslicer_model/Thu_Nov_16/11_01_52#30565#3/Metadata/pick_1.png" from 3mf Metadata/pick_1.png, ret 1

[... cut ...]
(process:30565): Gtk-CRITICAL **: 11:01:52.109: _gtk_style_provider_private_get_settings: assertion 'GTK_IS_STYLE_PROVIDER_PRIVATE (provider)' failed

(process:30565): Gtk-CRITICAL **: 11:01:52.109: _gtk_style_provider_private_get_settings: assertion 'GTK_IS_STYLE_PROVIDER_PRIVATE (provider)' failed

(process:30565): Gtk-CRITICAL **: 11:01:52.109: _gtk_style_provider_private_get_settings: assertion 'GTK_IS_STYLE_PROVIDER_PRIVATE (provider)' failed
Segmentation fault (core dumped)
RETURN 139
```

eropple, 2023-11-18:
Maybe related to #2714?

DiegoHerrera1890, 2024-02-14:
Hello Sir, 
I am new using OrcaSlicer and I'd like to run it using CLI commands, do you know where I can learn or read the commands? or how to use CLI? Thank you so much.

PR #2665 (2023-11-09): Fix cli segfault (#2661)
Partially fix #2661

The command in #2661 will still not work, due to missing the option `thumbnails_format`. Re-save the project using latest OrcaSlicer might solve that one.