OrcaSlicer/OrcaSlicer #2662: "Extra length on restart" setting is ignored

Smartnitz, 2023-11-08:
### OrcaSlicer Version

1.8 beta2

### OS version

Windows 11

### Printer

Bambu Lab P1P

### How to reproduce

1. Slice the Cube Primitive model
2. Change "Extra length on restart" value either in the "Printer>Extruder" Settings or the Filament>"Setting Overrides" Setting
3. Check the G-Code for any layer: In the very first line for each layer the extruder is supposed to unretract filament. Check the unretract length value in the G-Code

### Actual results

Whatever "Extra length on restart" is set to is ignored. It is only unretracting the regular retract length.
[image]

### Expected results

"Extra length on restart" value should be added to the regular unretract length.

### Project file & Debug log uploads

[Cube.zip](https://github.com/SoftFever/OrcaSlicer/files/13295753/Cube.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Smartnitz, 2023-11-08:
I just checked it in OrcaSlicer 1.7 and it has the same issue. Last version the feature was working is 1.6.6

Smartnitz, 2023-11-19:
The bug is still present in the final Version 1.8 :(

Smartnitz, 2023-11-28:
I noticed that this bug is related to the "wipe while retract" setting. Turning this off also fixes the issue with the "extra length on restart" setting. Still hoping for a fix soon, since both settings are really important for me

found the temporary fix here:
#https://github.com/SoftFever/OrcaSlicer/issues/2697

igiannakas, 2023-12-10:
Candidate fix here: https://github.com/SoftFever/OrcaSlicer/pull/3065 

please test and come back with observations.

PS. you shouldn't need extra length for restart with the P1P/S and your retraction value in the attached file is too high (0.8 is good for a P1X/X1C printer) but I assume you're doing that to demonstrate the bug.

Smartnitz, 2023-12-11:
The fix solves the problem and has worked perfectly for me so far.

Thank you so much <3

The setting is very practical for some filaments that virtually suck themselves back into the nozzle when retracted. 
I use 0.4 mm retraction + 0.1 mm extra. The values from the screenshot are for demonstration purposes only.

commit 0a31b33b (2023-12-13): Fix #2662 - "Extra length on restart" setting is ignored when wipe is enabled (#3065)

Fix #2662
closed 2023-12-13 by SoftFever