OrcaSlicer/OrcaSlicer #2663: Add Elegoo Neptune 4 Plus Printer Profile

sebastianbagya, 2023-11-08:
Please add elegoo neptune 4 plus printer profile, I already try to print on elegoo neptune 4 plus with elegoo neptune 4 max printer profile (adjusted size), but I see some error / warning in fluidd console. So I will be glad if the printer profile is officially added, thank you.

Axil94, 2023-11-12:
This would be amazing

CINCOxBINCO, 2023-11-13:
Need this

prisma7411, 2023-11-14:
würde mich auch interessieren :-), da es wirklich fehlt.

MathiasKrill, 2023-11-18:
same problem here, need this.

BlazS14, 2023-11-20:
This would help me as well :)

BarleyBrewhammer, 2023-11-22:
Need this :)

Badwolf7691, 2023-11-23:
Please, don't want top Patreon "Uncle Jessy" who Elegoo pays already.

AE3D-A99, 2023-11-23:
Thanks Badwolf7691

i need also the profile for the new neptune 4 + but i don´t want to buy a membership to get this from "uncle Jessy".
Especially not when elegoo is paying him.

GeneralDraga1701, 2023-11-27:
Please, By the gods of 3D Printing Add an Official Neptune 4 Plus profile! The Max profile with changed bed size doesnt seem to work right hinting at something else different with the Plus and Max.

GeneralDraga1701, 2023-11-27:
> Thanks @Badwolf7691 for the hint with that Uncle Jessy, i paid the 2$ to get a good working profile. Without that i would not be able to use orca - thank you ! UPDATE: That Uncle Jessys profil is trash for the printer everything i try out is trash.

how do you import a profile like uncle jessy to orca? Im still very newish.

BlazS14, 2023-11-27:
Just a psa for anyone using the n4 max profile on the n4+. I dont know why but i think that the fan structure on the backside of the y axis gantry isnt accounted for in the extruder size settings, so any prints that have the print by part setrring enabled (not by layer) may bump into previously printed parts on the bed. Otherwise the modified max profile with smaller bed dimensions works great for me.

w34sel, 2023-12-02:
Here i made a profile for the EN4P, try it out.
no guarantees :)
https://github.com/w34sel/3D-Printing/tree/main

vaxkun, 2023-12-03:
> Here i made a profile for the EN4P, try it out. no guarantees :) https://github.com/w34sel/3D-Printing/tree/main

Im waiting for the plus to arrive, is this working nice for u ppl?

CreativeLens, 2023-12-05:
> Here i made a profile for the EN4P, try it out. no guarantees :) https://github.com/w34sel/3D-Printing/tree/main

Hi,
Thanks @w34sel  for the Profile, I directly installed it.

One Question, and please excuse if it sounds a little dumb I am new to the Game, why are the Speedsettings so low? I get it for the first layer, but why are the other layers at 80 or 100? 
Till now I printed with the Neptune 3 Plus Profile and the settings were also very slow. Thought 250mm/s is recommended by Elegoo.

Can you tell me, what is best?

Kalooth, 2023-12-05:
@CreativeLens, the N4 Max profile that people usually say to copy and just change the bed size is running with those slower speeds. I've been running with my PR, https://github.com/SoftFever/OrcaSlicer/pull/2980  for over a week and haven't had any issues yet but... I'm a sample size of 1 so let me know if you give it a shot and if you run into issues.

w34sel, 2023-12-05:
@CreativeLens I chose that speed because it was the same for the N4pro and N4Max standard profile, so I thought it would be nice to maintain consistency by using the same values across the profiles. However, you can certainly use a higher speed if desired.

You can tweak it for each print.

Here is a small image marked with different colors:
Blue: Outer wall
Orange: Inner Wall
Purple Top: Surface
Red: Sparse Infill
White: Internal solid ( imagine this is fully printed)

[image]

francois-thi, 2023-12-09:
@CreativeLens Hello,

Brand new here in 3D printing ! Thanks for the profile, I downloaded it and installed it. I saw that the bed size is 330x330 in your profile, however the bed size for the Neptune 4 plus is 320x320. Is that normal or is that a mistake ? 

Thanks !

CreativeLens, 2023-12-09:
Hi,

I think it is a mistake, but the Profile is from @w34sel  :)

w34sel, 2023-12-09:
@francois-thi  I have copied the dimensions of the bed from the cura 4 elegoo profile for the n4plus

helmanis, 2023-12-20:
@francois-thi If you measure the bed manually then it will be 330x330. Was suprised too, but had to replace the PEI Plate, so i know ;)

francois-thi, 2023-12-20:
@helmanis Oh, thanks ! I didn't notice that :p

Shotgun-Monk, 2023-12-23:
Does this Orca profile work well? I am tired of the Elegoo Cura I am having issues that seem slicer related because if I print anything that Elegoo sliced like the Buddha or Flower pot it comes out perfect so it tells me the printer is mechanically sound.

Shotgun-Monk, 2023-12-24:
@w34sel, Thank You for making the Orca slicer profile. It is working great, better than the Uncle Jessy profile TBH.

w34sel, 2023-12-24:
Thx man :D
https://github.com/w34sel/3D-Printing/

francois-thi, 2023-12-30:
@Shotgun-Monk Hey, yes it's working great ;) I'm using it since a couple of weeks now, and I had no problems at all.

Georgelou35, 2024-01-16:
Can i make one question @w34sel?So what are the differenc between your profile and th stock neptune4 plus orca profile?🤔🤔

francois-thi, 2024-01-16:
@Georgelou35  Is there any Neptune 4 plus profile in Orca ?  I couldn't find one, only the Neptune 4 pro.

Georgelou35, 2024-01-16:
The new 1.9 has neptune 4 plus Profile

francois-thi, 2024-01-16:
@Georgelou35  Hurray ! Waiting for @w34sel answer then, but to be honest, it has been working perfectly fine for me since I downloaded it.

Georgelou35, 2024-01-16:
Also i wonder what are the differences between stock and w34sel profile

w34sel, 2024-01-16:
My bed size is 330x330 and the height is 387 (using dimensions from the Elegoo Cura profile) along with some speed settings. The current profile in version 1.9.0 was submitted by another guy who made his pull request earlier than mine.

Georgelou35, 2024-01-17:
W34sel thank you very very much.Now i will put to print a deadpool helmet to see the results.

w34sel, 2024-01-17:
@Georgelou35 please share your results with me :)

Georgelou35, 2024-01-17:
Yeah yeah.So far so good.when i go home in about 4 hours i will send picture here.

Georgelou35, 2024-01-17:
Also,do you know how i can when i transfer the gcode to transfer and the thumbnail???

Georgelou35, 2024-01-17:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Georgelou35, 2024-01-17:
Its going well.

w34sel, 2024-01-17:
@Georgelou35 nice :D 
Here is a repo for the thumbnails https://github.com/Molodos/ElegooNeptuneThumbnails-Prusa

Georgelou35, 2024-01-18:
So one dayhas pass.So far so good.I dont have so much strings the print is so clean.Can wait to finish.
[image]
[image]
[image]
[image]
[image]

Georgelou35, 2024-01-18:
Also before deadpool helmet i print a gengar with stock profile.But the results was not good🤷🤷 i will print it again with your profile when i will finish the deadpool helmet
[image]
[image]

[image]

w34sel, 2024-01-18:
@Georgelou35 Here for the thumbs https://github.com/fifonik/ElegooNeptuneThumbnailPrusaMod  
It is working fine.

Fleky, 2024-02-10:
> Gracias hombre: D https://github.com/w34sel/3D-Printing/

Muchas gracias amigo. Llevo una semana con orca y casi me di por vencido. Va muy bien.

w34sel, 2024-02-24:
Hey @Coop94,

i will create simple copy installer in the next few days for you and upload it on my Github repo. I will inform you when it is rdy.

Coop94, 2024-02-24:
Thanks w34sel.

Coop94, 2024-02-24:
Thank you so much. I’m new to all this. Sent from my iPhone

w34sel, 2024-02-24:
@Coop94  after rethinking it makes no sense for me to do this. 
Here a small instruction:
1) Download the repo as ZIP unpack and
2) go to §D-Printing/Elegoo Neptune 4 Plus/orca_19.0
3) copy the folders 2 folder to the following path in your home system (windows)
- copy "profiles" folder to "C:\Program Files\OrcaSlicer\resources" 
- copy "default" folder to "C:\Users\<USERNAME>\AppData\Roaming\OrcaSlicer\user"

Coop94, 2024-02-26:
If I don't have the second folder, do i just need to create one?

w34sel, 2024-02-26:
There should be one, if not drop the fault order to:
Type into your explorer %appdata% and go to "\OrcaSlicer\user" 
or
C:\Users\COOP94\AppData\RoamingOrcaSlicer\user

Coop94, 2024-02-26:
i found it. There was a check box to show hidden folders that wasn't checked. Thanks again

Kenepo17, 2024-04-13:
> There should be one, if not drop the default order to: Type into your explorer %appdata% and go to "\OrcaSlicer\user" or C:\Users\COOP94\AppData\RoamingOrcaSlicer\user

I did as instructed. I do use Elegoo Pla and Elegoo Rapid PETG. For the pla use your settings under sunlu pla?

closed 2024-07-21 by github-actions[bot]