OrcaSlicer/OrcaSlicer #2664: Vector too long FLsun V400

spmfred, 2023-11-08:
### OrcaSlicer Version

1.8.0 beta 2

### OS version

Windows 10 pro

### Printer

FlSun V400

### How to reproduce

Tried to slice a file and it goes through the slicing and at the end pops up and says "vector too long"

### Actual results

Won't output Gcode file

### Expected results

Should finish slicing and ask to save or send sliced gcode file
[christus.zip](https://github.com/SoftFever/OrcaSlicer/files/13297338/christus.zip)

### Project file & Debug log uploads

[christus.zip](https://github.com/SoftFever/OrcaSlicer/files/13297348/christus.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lightmaster, 2023-11-09:
I have received this error too, on a FLSun Q5. Error specifically is `cannot create std::vector larger than max_size()`

lightmaster, 2023-11-09:
Seems like this is probably the same as #2537, in which case it's reportedly fixed in a nightly build.

SoftFever, 2023-11-12:
It has been fixed in 1.8.0 RC2, can you check?

lightmaster, 2023-11-13:
I haven't seen it again on RC2, even on a project that was throwing that error every time before.

discip, 2023-11-18:
@spmfred
Have you tested the latest build?
If that solves it for you, please consider closing this issue. 😊

thanks

closed 2023-11-18 by SoftFever