OrcaSlicer/OrcaSlicer #2671: Tree Support boundaries not emmitted to Exclude Object?

grrminator, 2023-11-08:
### Discussed in https://github.com/SoftFever/OrcaSlicer/discussions/1328

<div type='discussions-op-text'>

<sup>Originally posted by **5teveO** June 11, 2023</sup>
Hi All.. 

I'll preface this by saying I am new to Orca and as such still learning. I am not new to 3d printing and generally use PS.. Orca has some great features and I am wanting to get along with it.

I had an issue last night,  Basic control box for my CNC, it has a few holes that need Tree supports just for a bit of help. I first encountered an issue as the linewidths were 0 - tree supports fail in that instance or did on this model (cant remember the error but thats incidental at the moment)

I'm running KAMP on klipper. II sliced the model - started the print in Fluidd.. and off it went. It was only then that I noticed that the KAMP Adaptive Purge line was located where i would expect it if no tree support was there, and then the tree support proceeded to print over the top of the purge line. The print still worked, so no major issues.. it just appeared that the tree supports addition to the model boundaries wasnt emiited in the polygon details for the exclude object.

I understand the skirt is ignored as that surround multiple objects, so wondering if the tree supports work in this way as they could be shared between objects?

Thank in advance

Steve</div>

Noisyfox, 2023-11-09:
Will take a look.

Noisyfox, 2023-11-09:
https://discord.com/channels/1137181739773603922/1150397562843582534/1169581246209343530

grrminator, 2023-11-09:
I am not in that discord so that link doesnt help me

But it may be a camp issue except it precicesly puts the purge line 10mm per default of the model every time, supports or not.

Is there not a fix aside from screwing with the CFG every time?

grrminator, 2023-11-09:
I see here. I will pass it on to the guy who can probably fix this for myself and his repo that tons of people use for these particular printers. Thank you!

discip, 2023-11-18:
@grrminator
If this is solved, please consider closing. 😊

discip, 2023-11-18:
@grrminator
If this is solved, please consider closing. 😊

thanks

StdVectorBool, 2024-02-18:
> Check the discussion here [discord.com/channels/1137181739773603922/1150397562843582534/1169581246209343530](https://discord.com/channels/1137181739773603922/1150397562843582534/1169581246209343530)
> 
> Basically this is an issue of KAMP, not Orca.

I wasn't able to load the Discord conversation.  What's the rationale behind considering this a KAMP issue?  It uses the boundaries provided in the gcode's Exclude objects list.

Before slicers supported it, Moonraker was using a geometry library to build an accurate boundary from the first layer of gcode:
https://docs.mainsail.xyz/overview/features/exclude-objects#enable-the-g-code-preprocessor-in-moonraker

but if the boundaries are already present in the gcode generated by the slicer then it trusts that despite it incorrectly excluding the footprint of non-STL objects like supports (like trees which can branch far from the object), prime towers, etc.

closed 2024-05-27 by github-actions[bot]