OrcaSlicer/OrcaSlicer #2674: Incorrectly shows a filament change when only using a filament besides the 1st filament and using a custom purge line in start gcode

lightmaster, 2023-11-09:
### OrcaSlicer Version

V1.8.0 Beta 2

### OS version

Pop_OS 22.04 LTS

### Printer

FLSun Q5

### How to reproduce

1. Have a custom purge line in your start gcode
2. Add 1 object to the plate
3. Set your object's filament to be something besides the 1st/default filament (ie. have Hatchbox as your 1st filament and Formfutura as your 2nd filament)
4. Set the 1 and only object on the plate to use the 2nd filament
5. Slice the plate

### Actual results

It shows that the custom purge line is using the 1st filament and the object to be printed is using the 2nd filament, and says there will be 1 filament change. However, if you export the gcode or look at it in the gcode viewer, there is no actual filament change. My filament change is using `M600` and manually switching filaments, however if I do a search on the resulting gcode, `M600` only shows up as a comment at the end of the file where it documents the settings used.

[image]

### Expected results

It should set the filament used for a custom purge line to the same filament that is the first to be printed as part of your objects, and if there is only 1 object and 1 filament used, it shouldn't show that a filament change is supposed to occur since there is no filament change code in the code.

ie:
- if you have 2 objects and have the plate set to print filament 2 first and then filament 1 second, then the custom purge line should show up as filament 2.
- if you have 1 object and it is set to use filament 2, then the custom purge line should show up as filament 2.

### Project file & Debug log uploads

[test-project.zip](https://github.com/SoftFever/OrcaSlicer/files/13305850/test-project.zip)
[log.zip](https://github.com/SoftFever/OrcaSlicer/files/13305851/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

olcayoren, 2023-11-09:
Hello, did you add m600 to the filament start section in the extruder machine gcode settings?
[image]

closed 2024-02-20 by github-actions[bot]

UTSquishy, 2025-04-30:
I hate to resurrect something this old, I'm likely just misunderstanding a setting, this is all new to me - but I'm having this same problem. It was working as expected a few days ago (but I changed a setting and borked everything and had to set things up new again), and now the purge line is extruder 1 when the whole print is extruder 2. Sovol SV02