OrcaSlicer/OrcaSlicer #2677: Display remaining print time for Flashforge printers

mariuske, 2023-11-09:
**Is your feature request related to a problem? Please describe.**
Add remaining print time for Flashforge printers when using Orca Slicer

**Which printers will be beneficial to this feature**
Flashforge Adventurer 5M and Flashforge Adventurer 5M Pro. Possibly other models as well.

**Describe the solution you'd like**
When using Orca Slicer as the slicer, the printer is only displaying current (elapsed) print time and not remaining print time (based on estimation given when slicing). It would be nice if remaining print time is displayed the same way as when using Flashprint.

**Describe alternatives you've considered**
Using Flashprint the remaining print time is displayed

@FlashforgeOfficial - Please look into this and let me know if I can assist in any way.

@SoftFever - This feature request can be assigned to @FlashforgeOfficial. Thank you.

FlashforgeOfficial, 2023-11-13:
Hi @mariuske , it would be good if we have this function fixed.  For now, it's still printing time elapsed displayed only.

mariuske, 2024-02-12:
@FlashforgeOfficial - Do you have an update on this?

FlashforgeOfficial, 2024-02-19:
> @FlashforgeOfficial - Do you have an update on this?

The request from #2811 is probably related to this.  So, I going to make another confirmation with Softfever.

closed 2024-05-27 by github-actions[bot]

JMcrafter26, 2024-07-10:
Is there any progress on this?

jcforbes57, 2024-12-14:
Also seeking a resolution to this

wsw70, 2024-12-22:
I have the same problem - but I do not even know where to look for the printing time elapsed (per @FlashforgeOfficial comment).
where is it displayed?

I only managed to have in the `Device` a feed from a camera (but as far as I can tell, this is just a frame for the HTTP stream)

jcforbes57, 2024-12-22:
> I have the same problem - but I do not even know where to look for the printing time elapsed (per @FlashforgeOfficial comment). where is it displayed?
> 

On the print status screen, like the main screen that's on while a print is running where it should the file name and the model being printed. The time elapsed is on the bottom right corner. With Flashprint and other slicers the same text is the time remaining instead of time elapsed.

wsw70, 2024-12-22:
> On the print status screen, like the main screen that's on while a print is running

Is this this screen?

[image]

I do not have (or I don't know I do) any screen that shows the printing process, except for the Device tab which is just the camera feed

jcforbes57, 2024-12-22:
No, we are talking about on the printer not on your computer.

wsw70, 2024-12-22:
> we are talking about on the printer not on your computer

Ahhhhh, sorry for that!

I will open a feature request then to get the information in Orca Slicer. It must be possible to query it since FlashPrinter shows it in the software.

jcforbes57, 2024-12-22:
Orca slicer contains the estimated print time already, it's on the slice page after it's done slicing.

wsw70, 2024-12-22:
> Orca slicer contains the estimated print time already, it's on the slice page after it's done slicing.

Yes, I know that - but I would like to have feedback from the printer. When sending something over the network there is the heat-up time and then the reality of progress.