OrcaSlicer/OrcaSlicer #2680: Unable to disable Gap Fill

gdsolar, 2023-11-10:
### OrcaSlicer Version

1.8.0 Beta 2

### OS version

Windows 11

### Additional system information

CPU: 12th gen i9-12900H
Memory: 16gb
GPU: RTX 3070 Ti

### Printer

Any

### How to reproduce

1. Set Gap Fill speed to 0 to disable Gap fill in Classic extrusion mode

### Actual results

An error box pops up that says, "Value is out of range." and automatically sets Gap Fill speed to 1.
[image]

### Expected results

The Gap Fill speed should be set to 0 and the feature should be disabled.

### Project file & Debug log uploads

[Project File.zip](https://github.com/SoftFever/OrcaSlicer/files/13314656/Project.File.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

nielsvz, 2023-11-11:
The minimum value for the speeds was changed in #2245

jeremytodd1, 2023-11-29:
I also agree that there needs to be a setting to disable gap fill.

Gap fill has the nasty side effect of just causing the toolhead to repeatedly poke at the top layer, which can cause quality issues due to stringing.

A simple setting to enable/disable gap fill would be great.

I'm aware of the "filter out tiny gaps" setting, but just fully disabling gap fill would also be a nice option to have.

closed 2024-01-18 by SoftFever