OrcaSlicer/OrcaSlicer #2682: 1.8beta2，桥接异常，平滑基础率导致异常速度

Dominguez-Z, 2023-11-10:
### OrcaSlicer Version

1.8.0-beta2

### OS version

Windows11

### Printer

creality CR6MAX

### How to reproduce

一、桥接异常
使用1.7.0正式版切片正常的桥接模型，在1.8测试版会在连接处欠缺长度，导致桥接不上，塌了。
随机使用默认的参数也是一样没接上。
使用工程文件中的1号盘测试。

二、平滑基础率导致异常速度
[image]
[image]

1、在开启熨烫和平滑挤出率的情况下，会出现速度只有1的极慢的层。
2、单独把熨烫关闭后，1速度的层就没有了，层速度的变化会不均匀，这个第三点应该是相关的。
3、换了单独把平滑挤出率关闭后，速度正常。

[image]

[image]

4、同样的模型在prusa中切出来较为均匀。
使用工程文件中的2号盘测试。

三、平滑挤出率对于同样的模型切片出来的结果不同
[image]
[image]
[image]
同样的方块模型9个，实心填充，开启平滑挤出率后，切片出来的速度变化并不是均一的。
1、接缝位置选择最近，每一个模型的实心填充起始点切片出来比较随机。变速也会在不同位置启动，但是是不均匀的启动，有些起始有变速，有些起始没有变速。
如果选择对齐，则会出现一层有变速一层没有。
2、实心填充开启或者关闭对结果基本不影响。
3、平滑挤出率仅在填充的起始有，收尾的时候没有，这个prusa的结果不一样。感觉prusa的比较合适。

### Actual results

一、桥接异常
1.8测试版切片结果
[image]
实际打印效果图
[image]

二、平滑基础率导致异常速度
1、极慢层
[image]

2、关闭熨烫，开启平滑挤出率
[image]

3、开启熨烫，关闭平滑挤出率
[image]

三、平滑挤出率对于同样的模型切片出来的结果不同
最近接缝的有些有有些没，中间的甚至是一直没有
[image]
[image]

对齐的
[image]
[image]

### Expected results

一、桥接异常
1.7.0正式版切片结果，有较长的一段在实体填充上，连接稳定。
[image]

二、平滑基础率导致异常速度
prusa中开启了压力均衡器，速度也是较为均匀的。
[image]

三、平滑挤出率对于同样的模型切片出来的结果不同
prusa的结果，起始有，收尾也有，且均一。
[image]

### Project file & Debug log uploads

[拉桥+平滑挤出率.zip](https://github.com/SoftFever/OrcaSlicer/files/13315408/%2B.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

igiannakas, 2023-11-10:
You will not see any smoothing if thr nozzle moves more than 2-3 mm from its last extrusion. The behaviour is different to prusa as arguably prusas implementation is slightly flawed. It doesn’t smooth on external perimeters and it smooths even after long moves , despite nozzle pressure having fallen to zero after a retraction move.

SoftFever, 2023-11-12:
@Dominguez-Z, thank you for reporting the issues. 
Regarding the Pressure Equalizer issue, please see @igiannakas's explanation. 

As for the bridge issue, it occurs in your test model because there's a thin layer that could be interpreted either as a top surface or a bridge surface. 
The differing behaviors between versions 1.7 and 1.8 are due to 1.8 incorporating some of the latest algorithms from PrusaSlicer. 
While ideally, this part should be considered a bridge rather than a top surface, such cases are very rare. It's very unlikely we will have single layer bridge in actual model.

closed 2024-02-20 by github-actions[bot]