OrcaSlicer/OrcaSlicer #2687: "Intelligent" segmented bed heating Neptune 4 Pro

Todoori, 2023-11-10:
**Is your feature request related to a problem? Please describe.**
No.
The Neptune 4 pro's heated bed is segmented.  To safe energy, you can only heat the inner part of the bed for smaller prints. This is not currently supported by the Orca Slicer (a workaround is available).

**Which printers will be beneficial to this feature**
Neptune 4 pro

**Describe the solution you'd like**
For the Neptune 4 pro the outer part of the bed can be turned off with the command *M141 S0*, therefore I have two possbile solutions I could think off (in Order not to disturb priming, M141 S0 should only be sent after priming.)

### Option 1
- Automatically determine whether or not the outer heating is needed, depending on the position of the print and it's size.
  - priming should not be considered 

### Option 2
- Before generating the Gcode, ask the user whether or not the outer bed is needed and add *M141 S0* to the Gcode if it is not needed
   - In Order for the user to determine whether or not the outer part is needed, a simple rectangle on the print bed would be enough.
[image]

**Describe alternatives you've considered**
I have created that Image I posted above and add or remove the line in my Startup Gcode depending on how large the print is.
This is scaled properly to fit on the neptune 4 pro preset in orca slicer. For everyone else wanting to use that workaround, feel free to import this image

[image]

HumbleDeer, 2023-11-12:
A third option is for you to make a printer profile or settings preset which incorporates the necessary gcode commands to either turn on the entire bed or turn on just the center. 

You can adjust the bed size of the profile with the smaller heating zone, and use home offsets. 
You can also adjust the bed keepout zone of the profile with the smaller heating zone, so the outer edges are out of bounds when using the profile with only the central part set up for heating.

HumbleDeer, 2023-11-12:
That's great work already! I'll have a look through the source code and see if I can find anything.
If I do, mind if I put this into a fork branch? I can put you in it as the author of the commits. It'd be good to get this in to a pull request so the other devs on here can chime in.

It might not be bad to have the outer boundaries as a variable if it is not one yet.

Todoori, 2023-11-13:
Thank you.
No, I don't mind but I don't think I can contribute much more than testing or giving feedback.

Yes, having the outer boundaries would be better than only having the model size and assuming it is in the middle of the bed.

discip, 2023-11-17:
@Todoori
@HumbleDeer

Maybe this could help:

boundaries:

X-min:   {first_layer_print_min[0]}
X-max:   {first_layer_print_max[0]}
Y-min:   {first_layer_print_min[1]}
Y_max:   {first_layer_print_max[1]}

discip, 2023-11-17:
@Todoori
@HumbleDeer

Maybe this could help:

boundaries:
```
X-min:   {first_layer_print_min[0]}
X-max:   {first_layer_print_max[0]}
Y-min:   {first_layer_print_min[1]}
Y_max:   {first_layer_print_max[1]}
```

HumbleDeer, 2023-11-18:
> @Todoori @HumbleDeer
> 
> Maybe this could help:
> 
> boundaries:
> 
> ```
> X-min:   {first_layer_print_min[0]}
> X-max:   {first_layer_print_max[0]}
> Y-min:   {first_layer_print_min[1]}
> Y_max:   {first_layer_print_max[1]}
> ```

I'll have a look into it, 't looks interesting and possibly useful. Cheers!

Todoori, 2023-11-24:
@HumbleDeer @discip 
I tried it with that and it seeems to get the correct Values (Terminal output via fluidd):
```
// Unknown command:"THRESHOLD= 5"
// Unknown command:"X_UPPER= 167.5 -$"
// Unknown command:"Y_LOWER= 57.5 +$"
// Unknown command:"Y_UPPER= 167.5 -$"
// Unknown command:"X_MIN= 3.94716"
// Unknown command:"X_MAX= 221.051"
// Unknown command:"Y_MIN= 3.94506"
// Unknown command:"Y_MAX= 221.045"
// Unknown command:"IF$"
```
But something else did not seem to work, no matter how large the print is, the outer bed get's turned off.
I figured my syntax is wrong. So i tried to stay as close to the Syntax I could find in the end Gcode

#### This is now at the end of my start Gcode:
```
;Depending of the size of the print, only turn on the inner bed

M141 S0 ; Turn off outer part of the bed	 
{if first_layer_print_min[0] < 62.5}M141 S[bed_temperature_initial_layer_single] ; Turn on outer part of the bed if Xmin < 57.5 + Threshold {endif}
{if first_layer_print_max[0] > 162.5}M141 S[bed_temperature_initial_layer_single]; Turn on outer part of the bed if Xax > 167.5 - Threshold {endif}
{if first_layer_print_min[1] < 62.5}M141 S[bed_temperature_initial_layer_single]; Turn on outer part of the bed if Ymin < 57.5 + Threshold {endif}
{if first_layer_print_max[1] > 162.5}M141 S[bed_temperature_initial_layer_single] ; Turn on outer part of the bed if Ymax > 167.5 - Threshold {endif}
```

#### And before layer change:
```
;Depending of the size of the print, only turn on the inner bed
M141 S0 ; Turn off outer part of the bed	 
{if first_layer_print_min[0] < 62.5}M141 S[bed_temperature]; Turn on outer part of the bed if Xmin < 57.5 + Threshold {endif}
{if first_layer_print_max[0] > 162.5}M141 S[bed_temperature] ; Turn on outer part of the bed if Xax > 167.5 - Threshold {endif}
{if first_layer_print_min[1] < 62.5}M141 S[bed_temperature] ; Turn on outer part of the bed if Ymin < 57.5 + Threshold {endif}
{if first_layer_print_max[1] > 162.5}M141 S[bed_temperature] ; Turn on outer part of the bed if Ymax > 167.5 - Threshold {endif}
	 
```

## Not very pretty, but seems to work:

#### Large Print Gcode Output:

Start
```

;Depending of the size of the print, only turn on the inner bed

M141 S0 ; Turn off outer part of the bed	 
M141 S60 ; Turn on outer part of the bed if Xmin < 57.5 + Threshold 
M141 S60; Turn on outer part of the bed if Xax > 167.5 - Threshold 
M141 S60; Turn on outer part of the bed if Ymin < 57.5 + Threshold 
M141 S60 ; Turn on outer part of the bed if Ymax > 167.5 - Threshold 

```

Layer change
```
;LAYER_CHANGE
;Z:0.6
;HEIGHT:0.2
;BEFORE_LAYER_CHANGE
G92 E0

;Depending of the size of the print, only turn on the inner bed
M141 S0 ; Turn off outer part of the bed	 
M141 S60; Turn on outer part of the bed if Xmin < 57.5 + Threshold 
M141 S60 ; Turn on outer part of the bed if Xax > 167.5 - Threshold 
M141 S60 ; Turn on outer part of the bed if Ymin < 57.5 + Threshold 
M141 S60 ; Turn on outer part of the bed if Ymax > 167.5 - Threshold 

```

#### Small Print Gcode Output:

Start
```
;Depending of the size of the print, only turn on the inner bed
M141 S0 ; Turn off outer part of the bed	 

```

Layer change
```
;LAYER_CHANGE
;Z:0.2
;HEIGHT:0.2
;BEFORE_LAYER_CHANGE
G92 E0

;Depending of the size of the print, only turn on the inner bed
M141 S0 ; Turn off outer part of the bed	 

```

Still needs testing but I am quite confident with that result already.

If anybody knows how to improve the syntax with variables or and's in the if statements I'm happy to include that.
Thank you for your help already!

closed 2024-03-01 by github-actions[bot]

danielschwab, 2024-03-11:
I simplified / improved your code a little bit:
```
{if first_layer_print_min[0] >= 62.5 && first_layer_print_max[0] <= 162.5 && first_layer_print_min[1] >= 62.5 && first_layer_print_max[1] <= 162.5}M141 S0 ; Turn off outer part of the bed {endif}
```

This prevents the on\off cycle in the resulting g-code.

I just added it to the **after layer g-code** of OrchaSlicer because in the **start g-code** the print bed is heated up completely which seems to be important for the purge line at the top of the bed. After this the outer bed heater is turned off if the printing area is small enough.

Sometimes there is a `M140 S60 ; set bed temperature` statement in the g-code which reactivates the outer bed heater but after the next layer change, the outer bed heater is deactivated again. In my tests this statement only occurs once.

This behavior seems to sufficient to save electricity for longer prints with a small printing area.

maj429, 2024-07-15:
[image]
Neptune 4 Pro here.
I used the code posted by Daniel, and I checked my gcode, there is no M140 command present to turn on the outer bed per layer, BUT it's not staying off. It turns off each layer, then immidatly back on, and it's not the Gcode. I even tried turning off bed1 from the fluidd interface, manually setting it to 0, and it won't stay more than a fraction of a second, before reverting to full temp (75c for the print I screenshotted.) Any reason the printer won't let me change the bed temps via this code or even manually mid print?

trueserve, 2024-07-19:
@maj429 This worked with older versions of Orca Slicer. The method in @danielschwab's post broke some time at or before release 2.1.1. Inspecting the gcode, the outside heater is turned off after layer change section as expected, but the bed temps are then set immediately after with M140 right below `;_SET_FAN_SPEED_CHANGING_LAYER`

I haven't looked into the cause.

One workaround would be 1) editing `printer.cfg` to only operate the center bed element on M140 and 2) changing startup code to send M140+M141 for bed temperatures. Anything else that assumes M140 operates both elements (possibly the LCD-driven bed mesh, what else?) may break though.

J0J0-D1D10, 2024-09-19:
any developement about the issue?

Mattis3dprinting, 2024-10-12:
I was having the same issue as all of you. Tinkering with it today I was able to take Daniel's code and modify it slightly replacing my M190 command in the start gcode with the one shown below. 

I did have to go in and modify my klipper config file so that the M191 macro was not the same as the M190 macro. I deleted the line that said SET_HEATER_TEMPERATURE HEATER=heater_bed1 TARGET={TARGET|int} in my M191 macro so that it would only heat the inner bed instead of both.

After that paste this in place of the M190 in that start gcode and have you extra code in the before layer change that was originally posted and this should work.

```
{if first_layer_print_min[0] >= 62.5 && first_layer_print_max[0] <= 162.5 && first_layer_print_min[1] >= 62.5 && first_layer_print_max[1] <= 162.5}M191 S[bed_temperature_initial_layer_single] ; Turn off outer part of the bed {endif}
{if first_layer_print_min[0] < 62.5 && first_layer_print_max[0] > 162.5 && first_layer_print_min[1] < 62.5 && first_layer_print_max[1] > 162.5}M190 S[bed_temperature_initial_layer_single] ; Turn on outer part of the bed {endif}
```

v-2841, 2024-11-23:
I wrote to Elegoo support about this issue. They sent me the following answer:

> When the heatbed is set to 70°C or above, the external heatbed is automatically enabled. This behavior is designed to ensure stable temperature control and prevent fluctuations that could affect print quality. Unfortunately, this feature is hard-coded into the software, and it cannot be switched off or changed through standard user settings.

So there is no problem in the slicer. You just need to set the temperature below 70 °С, and change the temperature of the subsequent layers to 0, so that the M140 command is not entered additionally after the first layer.

[image]

And I added @danielschwab code in start g-code:

[image]