OrcaSlicer/OrcaSlicer #2697: retraction amount not added /extruded after retraction

lynara, 2023-11-11:
### OrcaSlicer Version

1.8.0-rc

### OS version

Windows 11

### Printer

Voron 2.4

### How to reproduce

Slice model and and look at code where a retraction is happening.

### Actual results

Filament is retracted by specified amount but when continuing printing it is no pushed back / extruded by the retracted amount.
Printing continues as if no retraction had happend.
[image]

### Expected results

Filament is retracted by specified amount. Before continuing printing it is pushed back / extruded by the retracted amount like in 1.7.0
[image]

### Project file & Debug log uploads

[Box - Bottom.zip](https://github.com/SoftFever/OrcaSlicer/files/13325599/Box.-.Bottom.zip)
[debug_Sat_Nov_11_14_15_09_36368.log.0.zip](https://github.com/SoftFever/OrcaSlicer/files/13325603/debug_Sat_Nov_11_14_15_09_36368.log.0.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

hliebscher, 2023-11-11:
yes ist is a bug   if you add wipe while retracting in filament settings .... fix this
pls add under others in profile
   add numbers and verbose G-Code than you can easy find 

N580 G1 E-1.05 ; **retract**
N581 ;WIPE_START
N582 G1 X187.243 Y189.653 E-.45 ; **wipe and retract**
N583 ;WIPE_END
N584 G1 X246.961 Y119.516 F21000
N585 G1 E1.5 F1800 ;  ; **unretract**
N586 ;TYPE:Bottom surface
N587 ;WIDTH:0.720173

closed 2023-11-12 by SoftFever

SoftFever, 2023-11-12:
Thank you.
This bug has been fixed in 1.8.0 RC2

PR #2699 (2023-11-12): Fix a regression issue that retraction is not correct when wipe is off
Fix #2697
commit 15358b21 (2023-11-12): Fix a regression issue that retraction is not correct when wipe is off
Fix #2697