OrcaSlicer/OrcaSlicer #2698: Anchor Solid Infill by X mm from SuperSlicer

djngnear, 2023-11-11:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

The top layer is often left with holes when printing text on top of a flat object. I'd like to have a nice clean monotonic top layer underneath the text.

[image]
[image]

**Which printers will be beneficial to this feature**
All printers

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

I love the SuperSliceroption to fill in small areas with solid infill on top surfaces like under text or smaller objects. I went to Superslicer for this very reason. By default it's set to 150% but I like to bump it up to 300 or 400%. That way the monotonic infill just looks better and it doesn't need to navigate around holes.
[image]
[image]

rizwansarwar, 2023-11-19:
+1 

This although some cases slightly more filament, saves a lot of time in printing, gives better surface finish and reduces chances of bad surfaces due to poor bridging. I think this should really be implemented as an option.

djngnear, 2023-11-19:
> +1
> 
> This although some cases slightly more filament, saves a lot of time in printing, gives better surface finish and reduces chances of bad surfaces due to poor bridging. I think this should really be implemented as an option.

I agree. I make a lot of signs and it really affects my top layers.

rizwansarwar, 2023-11-28:
[image]

@SoftFever Another example, when printing a Lego brick. There is unnecessary holes on top surface, the studs of the lego start above the top surface, so I am not sure if it really necessary to have these holes on top surface and have the algorithm fill around those holes. An option to fill to not do it would be really useful. This is present in Super Slicer where infill can be anchored by X amount on certain layer heights to eliminate this type of stuff.

It is an advanced user option but extremely useful to have.

rizwansarwar, 2024-02-29:
Why is this marked as stale? this is a valid request albiet low priority.

@SoftFever Do you think this is worth adding to list of wants?

closed 2024-06-07 by github-actions[bot]

TheBoojah, 2025-12-07:
This setting is really useful in a wide number of instances.
Note the "bridged" option on the same line of the first screenshot in SuSi, this option is very useful as a lot of internal bridges become much easier to print cleanly, and where disjoint solid infills creates weak spots in the printed geometry.